PSP-Archive/ARK-4 #223: psp slow to start

freelancers19, 2023-09-04:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.63 Rev 2

### Homebrew/GAME (if applicable)

noumd

### Plugins ( if applicable )

vhs, ms0:/seplugins/noumd.prx, on
game, ms0:/seplugins/noumd.prx, on

### Steps to reproduce

PSP is slow to start up. I want to disable reading from the UMD, I tried to install the NoUMD plugin, but it doesn't work on Ark-4 4.20.63 Rev 2. How can I fix it?

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

freelancers19, 2023-09-05:
I have solved. I was wrong to write "VSH"
it works

krazynez, 2023-09-05:
Okay good thanks for the update.

closed 2023-09-05 by krazynez