PSP-Archive/ARK-4 #224: Text spacing in Network menu (ARK CL)

Jambul46, 2023-09-06:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

Latest*

### Homebrew/GAME (if applicable)

ARK Custom Launcher

### Steps to reproduce

Open Network menu in CL and overfill text window 
The last line goes under the screen

### USB Charge

Always

### Overclock

Homebrew

### Power Save

Disabled

### Balanced Energy Mode

Game

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Alway

### Inferno Cache

Off

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-09-06:
Yes, he's probably using FTP. This issue may have come from adding the "Press [] To update" line without adjusting the maximum amount of lines it can display. So if it had a maximum of 10, it'll draw 11 instead.

JoseAaronLopezGarcia, 2023-09-06:
Should be fixed in latest pre-release.

closed 2023-09-06 by Jambul46