PSP-Archive/ARK-4 #225: ARK installed via FastARK sometimes freezes the PS TV

Goldtuga, 2023-09-08:
### PSP/Vita Model

VTE-1000 (Vita TV)

### ARK Version

4.20.63 Rev 2

### Steps to reproduce

ARK installed via FastARK sometimes freezes and I have to turn off the PS TV (including being forced to press its power button because the PS TV doesn't respond to the controller).

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-09-08:
Where exactly does it freeze? During loading of the CFW? On the Custom Launcher? On games?
I need a bit more information.

Goldtuga, 2023-09-08:
> Where exactly does it freeze? During loading of the CFW? On the Custom Launcher? On games? I need a bit more information.

Sorry. I mean right after running ARK. What's weird is that apparently it only freezes once right after being installed and then it works fine. Also, ARK installed by NoPspEmuDrm doesn't freeze, although it's more complex so should I use the latter when installing standalone ARK?

JoseAaronLopezGarcia, 2023-09-08:
> > Where exactly does it freeze? During loading of the CFW? On the Custom Launcher? On games? I need a bit more information.
> 
> Sorry, I meant right after running ARK. What's weird is that apparently it only freezes once right after being installed and ran and then it works fine. Also, ARK installed by NoPspEmuDrm doesn't freeze, although it's more complex, so should I use the latter when installing standalone ARK?

It's possible that you're using an outdated `K.BIN` file (found inside `ARK_01234`). Older ones had stability issues and would often crash. Newer ones are more stable and haven't caused crashes in a while. The thing is that the updater doesn't touch that file, you have to manually update it yourself.
I'm not entire sure that FastARK has the updated `K.BIN`, but we will update FastARK soon once we have the new ARK release ready.

Goldtuga, 2023-09-08:
> It's possible that you're using an outdated `K.BIN` file (found inside `ARK_01234`). Older ones had stability issues and would often crash. Newer ones are more stable and haven't caused crashes in a [...]

But I've manually updating myself so I'm not using the outdated K.BIN file and one thing is the kernel error which I stopped having with the latest version and another is this freezing issue where I have to turn off the PS TV by force (I can exit ARK fine after having the kernel error but not when having this freezing issue since the PS TV doesn't respond to the controller and I have to press its power button until it turns off or take its power cable). 

At least I don't have the kernel error and the freezing issues with ARK installed by FastARK.

JoseAaronLopezGarcia, 2023-09-08:
So the freezing issues are in the Menu?

Goldtuga, 2023-09-08:
> So the freezing issues are in the Menu?

Even before it appears. It freezes on a black screen.

Goldtuga, 2023-09-08:
> So the freezing issues are in the Menu?

Sometimes, even before it appears. It randomly freezes.

If ARK only has one version, then why does the one installed via FastARK crashes unlike the one installed by NoPspEmuDrm?

EDIT: Now it froze in the "Warning:" menu, right after the "minis" logo showed up.

JoseAaronLopezGarcia, 2023-09-08:
The one installed by FastARK might be outdated.

krazynez, 2023-09-08:
> The one installed by FastARK might be outdated.

Updated to latest release (again) :wink:

Goldtuga, 2023-09-08:
> The one installed by FastARK might be outdated.

It was the 4.20.63 Rev 2, so it was the lattest at the time.

> > The one installed by FastARK might be outdated.
> 
> Updated to latest release (again) 😉

How, since the latest version is 4.20.64 Rev 1 and FastARK installs the 4.20.64 version?

JoseAaronLopezGarcia, 2023-09-08:
Rev 1 was just released, you can update using Network Tools.

krazynez, 2023-09-08:
I just updated FastARK-4 to rev 1 as well. But yes please just from now on use the Network Tools to update to the latest Stable version of ARK.

Goldtuga, 2023-09-08:
> Rev 1 was just released, you can update using Network Tools.

> I just updated FastARK-4 to rev 1 as well. But yes please just from now on use the Network Tools to update to the latest Stable version of ARK.

This time, before installing ARK using FastARK, I refreshed the database, and I didn't have this issue. Also, after removing ARK, all of its folders were removed as well, unlike before.

Goldtuga, 2023-09-08:
> Rev 1 was just released, you can update using Network Tools.

> I just updated FastARK-4 to rev 1 as well. But yes please just from now on use the Network Tools to update to the latest Stable version of ARK.

This time, before installing ARK using FastARK, I refreshed the database, and I didn't have this issue. Also, after removing ARK, all of its folders were removed as well, unlike before.

EDIT Will the PSP's and PS Vita's Custom Launcher and PSP's XMB system update always update ARK to the latest version, including the latest Rev version or the version with the latest commits?

JoseAaronLopezGarcia, 2023-09-08:
> EDIT Will the PSP's and PS Vita's Custom Launcher and PSP's XMB system update always update ARK to the latest version, including the latest Rev version or the version with the latest commits?

There's a file called UPDATER.TXT in ARK_01234, this file points to the update server URL. The one shipped by default only updates to latest release, but not pre-release. If you want to update to latest pre-release you have to use the test server by simply adding /test at the end of the updater URL.

Goldtuga, 2023-09-08:
> There's a file called UPDATER.TXT in ARK_01234, this file points to the update server URL. The one shipped by default only updates to latest release, but not pre-release. If you want to update to [...]

The versions I was talking about were non-pre-release ones since they often have Rev versions and the same version even has commits almost every day.

krazynez, 2023-09-08:
> The versions I was talking about were non-pre-release ones since they often have Rev versions and the same version even has commits almost every day.

Rev's are Stable releases still. Just the GitHub Main repo is test branch (/test) ever push we send to git will auto sync to be the latest /test we don't bump the version unless a release/rev is happening

Goldtuga, 2023-09-08:
> Rev's are Stable releases still. Just the GitHub Main repo is test branch (/test) ever push we send to git will auto sync to be the latest /test we don't bump the version unless a release/rev is [...]

But in my case it wasn't updating to the latest version when a Rev one was available and in some cases it didn't update to the latest non-Rev version and in one case it downgraded from a Rev 1 version to the previous non-Rev version. My question is, when updating, will Custom Launcher and the XMB install the version with the latest commits? And what if enabling the "Force update" setting, will it reinstall the same version if at the time it's the latest one but with the latest commits?

JoseAaronLopezGarcia, 2023-09-08:
Like I said, the only way to update to the latest commit (pre-release) is to use the `/test` server. The regular server will always only have the latest regular release.

krazynez, 2023-09-08:
Latest Stable:
UPDATER.txt consist `http://ark-4.ddns.net`

if `Force Update` is enabled it will reinstall the Latest Stable as of right now (4.20.64 r1)

Latest commit (/test) "Bleeding edge":
UPDATER.txt consist `http://ark-4.ddns.net/test`

if `Force Update` is enabled it will (re)install the bleeding edge (commit), do not reference the version number because we do not update it for /test

Is that more clear?

Goldtuga, 2023-09-08:
> Like I said, the only way to update to the latest commit (pre-release) is to use the `/test` server. The regular server will always only have the latest regular release.

OK, but like I said, in my case there was a Rev version available and it didn't install it and in another case it actually downgraded from the Rev 1 version to the non-Rev version when the Rev 2 version was available.

JoseAaronLopezGarcia, 2023-09-08:
Because you are not using the `/test` server, so you will only have regular releases available.

Goldtuga, 2023-09-08:
> Because you are not using the `/test` server, so you will only have regular releases available.

Then why did it update to the 4.20.64 version to the Rev 1 just know? I was using that server, then stopped using it and now I am somehow using it again?

JoseAaronLopezGarcia, 2023-09-08:
Because Rev 1 is now the latest release so it's available on the regular server.

Goldtuga, 2023-09-08:
> Because Rev 1 is now the latest release so it's available on the regular server.

So, the other times it didn't update to the latest Rev version, they weren't on the regular server? Also, what about when the Custom Launcher actually downgraded ARK?

JoseAaronLopezGarcia, 2023-09-08:
> > Because Rev 1 is now the latest release so it's available on the regular server.
> 
> So, the other times it didn't update to the latest Rev version, they weren't on the regular server? Also, what about when the Custom Launcher actually downgraded ARK?

You probably manually installed a pre-release then used the custom launcher with "forced update" enabled and it downgraded your version with the regular release.
I don't know exactly what you did I am only making theories. In any case, it doesn't matter. I don't recommend installing pre-release versions if you're not a tester, best to stick to stable releases.

Goldtuga, 2023-09-08:
> You probably manually installed a pre-release then used the custom launcher with "forced update" enabled and it downgraded your version with the regular release. I don't know exactly what you did I [...]

I never installed pre-releases, only when I was told to see if I still had an issue in ARK and that was last week.

JoseAaronLopezGarcia, 2023-09-08:
Alright good. So is are the freezing issues solved? Can I close this thread?

Goldtuga, 2023-09-08:
> Alright good. So is are the freezing issues solved? Can I close this thread?

I also had this issue when updating from the XMB, so the "Force dowload" Custom Launcher setting doesn't apply here, and like I stated, I also had this issue when updating from a non-Rev version to another one, so both were non-pre-release versions. The XMB even told there were no new versions available when they were.

No, let me test it further.

Goldtuga, 2023-09-09:
Now I'm not having the issue but when removing ARK, its folders aren't being deleted again.

closed 2023-09-09 by JoseAaronLopezGarcia

Goldtuga, 2023-09-10:
Why doesn't ARK's GitHub mention the FastARK and ChovySign methods unlike ARK's GitHub's Wiki?

krazynez, 2023-09-10:
You will see why soon.

Goldtuga, 2023-09-10:
> You will see why soon.

Could you state in ARK's GitHub and ARK's GitHub's Wiki under "Full Flash Installation" this step can only be done after doing one of the three permanent ARK install methods?

JoseAaronLopezGarcia, 2023-09-10:
> > You will see why soon.
> 
> Could you state in ARK's GitHub and ARK's GitHub's Wiki under "Full Flash Installation" this step can only be done after doing one of the three permanent ARK install methods?

But that's not true at all, you can do Full Installation on Live CFW mode.

Goldtuga, 2023-09-10:
> > > You will see why soon.
> > 
> > 
> > Could you state in ARK's GitHub and ARK's GitHub's Wiki under "Full Flash Installation" this step can only be done after doing one of the three permanent ARK install methods?
> 
> But that's not true at all, you can do Full Installation on Live CFW mode.

ARK's GitHub's Wiki doesn't state one can have stability issues after running the standalone PS Vita ARK via FastARK method unlike ARK's GitHub (how can one run the "freemem ()" alghoritm?), doesn't mention ARK has limitations with PS1 games unlike ARK's GitHub and ARK's GitHub states "ARK CFW" must be run after running Infinity the second time instead of "ARK Loader" so could you at least state the same in both places and replace "ARK CFW" by "ARK Loader" under ARK's Github "Permanent CFW via Infinity" step?

JoseAaronLopezGarcia, 2023-09-10:
> ARK's GitHub's Wiki doesn't state one can have stability issues after running the standalone PS Vita ARK via FastARK method unlike ARK's GitHub (how can one run the "freemem ()" alghoritm?), doesn't [...]

You're getting confused with a part of the readme that is for `Developers` to make their own `K.BIN` (kernel exploit file), not for users.

Goldtuga, 2023-09-10:
> You're getting confused with a part of the readme that is for `Developers` to make their own `K.BIN` (kernel exploit file), not for users.

It isn't mentioned it's for developers (it's only mentioned it's needed if one is running ARK on an ancient FW) as well as what is the "freemem ()" alghoritm and the stability issue isn't mentioned at all in ARK's GitHub's Wiki. 

Also, what about the rest (ARK's GitHub's Wiki not mentioning ARK has limitations with PS1 games unlike ARK's GitHub and ARK's GitHub stating "ARK CFW" must be run after running Infinity the second time instead of "ARK Loader")?

krazynez, 2023-09-10:
I have told you this mutliple times the Wiki is not finalized.

Goldtuga, 2023-09-10:
> I have told you this mutliple times the Wiki is not finalized.

Then it shouldn't exist if it's not ready to not confuse users. It has information the main page doesn't and it even contradicts it. Also, I'm complaining about the main page as well, not only about the Wiki.

JoseAaronLopezGarcia, 2023-09-10:
If it doesn't exists, then how are we gonna work on it and improve it?
We work on this on our free time, this isn't a full time job for a company that pays us to do this or do that, we try to be as professional as possible but we all have our lives to live.
A project so big as this one doesn't just come out of nowhere with perfect code and perfect documentation. Stuff gets improved as we go along.
Just going over the wiki to find mistakes is a hugely time-consuming thing, not something you can do in 5 minutes.

Goldtuga, 2023-09-10:
> If it doesn't exists, then how are we gonna work on it and improve it? We work on this on our free time, this isn't a full time job for a company that pays us to do this or do that, we try to be as [...]

By keeping just one version (the main page one).

I understand that but there are things that are contradictory and so end up confusing users.

JoseAaronLopezGarcia, 2023-09-10:
Our aim is to simplify the main page and populate the wiki. Hence why the readme now only has two sections for PS Vita (Standalone and Adrenaline) but the wiki has more sections (i.e. Standalone via ChovySign).
One should be a summary, the other should be fully detailed.

Goldtuga, 2023-09-10:
> Our aim is to simplify the main page and populate the wiki. Hence why the readme now only has two sections for PS Vita (Standalone and Adrenaline) but the wiki has more sections (i.e. Standalone via ChovySign). One should be a summary, the other should be fully detailed.

But in some cases the main page has more information than the Wiki.

JoseAaronLopezGarcia, 2023-09-10:
Because the wiki is rather new so for a very long time the readme was the only place that got any information.

Goldtuga, 2023-09-10:
> Because the wiki is rather new so for a very long time the readme was the only place that got any information.

Could you update the Wiki regarding the new NoPspEmuDrm instructions?

Also, it's confusing having 3 methods to install the standalone PS Vita ARK (why isn't 1 enough? Before, there was only the FastARK method.) as well as 3 methods to update the PSP ARK on ARK's GitHub. Why isn't mentioned in ARK's GitHub how to update the standalone and Adrenaline ARK and in ARK's GitHub's Wiki how to update the PSP ARK, the standalone PS Vita ARK via the NoPspEmuDrm and ChovySign methods and the Adrenaline ARK? Why does Adrenaline ARK even exist if the standalone PS Vita ARK already exists?

JoseAaronLopezGarcia, 2023-09-10:
Adrenaline has more features than Standalone, such as the XMB and better PS1 game support.

Goldtuga, 2023-09-10:
> Adrenaline has more features than Standalone, such as the XMB and better PS1 game support.

Then that should be stated. Why do 3 differents methods to install the PS Vita standalone ARK exist?

JoseAaronLopezGarcia, 2023-09-10:
> > Adrenaline has more features than Standalone, such as the XMB and better PS1 game support.
> 
> Then that should be stated. Why do 3 differents methods to install the PS Vita standalone ARK exist?

It IS very CLEARLY stated that Standalone has less features and limitations.

[image]

Goldtuga, 2023-09-10:
> Why do 3 differents methods to install the PS Vita standalone ARK exist?

OK, but why do 3 differents methods to install the PS Vita standalone ARK exist and could you update the Wiki regarding the new NoPspEmuDrm instructions?

JoseAaronLopezGarcia, 2023-09-10:
If you check the readme, we've simplified it down to only one method (FasterARK). However, more advanced users that wish to create their own ARK Loader bubbles with ChovySign can still see the steps needed in the wiki.

Goldtuga, 2023-09-10:
> If you check the readme, we've simplified it down to only one method (FasterARK). However, more advanced users that wish to create their own ARK Loader bubbles with ChovySign can still see the steps needed in the wiki.

I already checked but now ARK's GitHub Wiki doesn't mention the standalone PS Vita ARK via NoPspEmuDrm unlike ARK's GitHub.

JoseAaronLopezGarcia, 2023-09-10:
Because the NoPspEmuDrm method has been merged with the FasterARK method and now they are part of the same process (a much easier and simplified process).

Goldtuga, 2023-09-10:
> Because the NoPspEmuDrm method has been merged with the FasterARK method and now they are part of the same process (a much easier and simplified process).

So, now there are only 2 methods do install the standalone PS Vita ARK (FasterARK and ChovySign)?

JoseAaronLopezGarcia, 2023-09-10:
> > Because the NoPspEmuDrm method has been merged with the FasterARK method and now they are part of the same process (a much easier and simplified process).
> 
> So, now there are only 2 methods do install the standalone PS Vita ARK (FasterARK and ChovySign)?

Yes but ChovySign is for more advanced users. FasterARK is the main method.

Goldtuga, 2023-09-10:
> Yes but ChovySign is for more advanced users. FasterARK is the main method.

Maybe you should get a different name since this one combines both FasterARK and NoPspEmuDrm ones and mention it in both the main page and Wiki because calling it FasterARK is weird.

JoseAaronLopezGarcia, 2023-09-10:
NoPspEmuDrm is optional (as it CLEARLY states in the readme AND wiki). I think you are looking too much into an issue that isn't there. I don't any more time to reply unless it's something important so I'll leave this thread as is.

Goldtuga, 2023-09-10:
> NoPspEmuDrm is optional (as it CLEARLY states in the readme AND wiki). I think you are looking too much into an issue that isn't there. I don't any more time to reply unless it's something important so I'll leave this thread as is.

It states it's recommended, not optional.