PSP-Archive/ARK-4 #226: CWCheatpops Not loading cheats

ImpactKaizer, 2023-09-08:
So I wanted to ask first since I just switched from ME CFW and wanted to give this 
CFW a shot (was mostly curious).
The Usual plugins I use seem fine for the most part expect for it not loading CwCheatsPop cheat codes it loads the CWcheats menu but shows mostly blank , do you need to put the CWCheatPOP.DB in a specific location rather than the Usual spot ?

The device I'm using is a PSP-GO (N1000) for reference as well 
Game used was Super Robot Taisen F (SLPS_01727)
And this is within the Plugins.txt
game, ef0:/seplugins/cwcheat/cwcheat.prx, on
xmb, ef0:/seplugins/cxmb/cxmb.prx, on
ps1, ef0:/seplugins/CWCHEAT/cwcheatpops.prx, on

[image]

krazynez, 2023-09-08:
Related https://github.com/PSP-Archive/ARK-4/issues/216

JoseAaronLopezGarcia, 2023-09-08:
This just means the cheat database is not updated. Most cwcheat downloads contain a very small and outdated database.
For reference, the most up to date cheat.db must be about 8MB or more, while most downloads contain a database that is only 2MB in size.

For more information (and solution), you can go here:
https://github.com/PSP-Archive/ARK-4/issues/153

ImpactKaizer, 2023-09-08:
> This just means the cheat database is not updated. Most cwcheat downloads contain a very small and outdated database.
> For reference, the most up to date cheat.db must be about 8MB or more, while [...]

It use to work fine with ME cfw , I'll give it a Shot and see worst case I'll rollback if won't play ball with PS1 cheats .

Thanks for the reply And on the matter cheers .

JoseAaronLopezGarcia, 2023-09-08:
I've found that the issue is somehow related to "Memory Stick Speedup", disabling that setting makes cwcheat work fine.
I am currently looking into a proper fix.

JoseAaronLopezGarcia, 2023-09-08:
So this is also an issue with `PRO`, however it's not an issue with `ME`.

JoseAaronLopezGarcia, 2023-09-08:
Latest pre-release should fix this.

JoseAaronLopezGarcia, 2023-09-08:
Fixed in latest release.

closed 2023-09-08 by JoseAaronLopezGarcia

ImpactKaizer, 2023-09-08:
> Fixed in latest release.

Yeah I tried the New Build since I'm migrating my 3000 and the Cheatpops loads fine minor nitpick (regarding the background showing the game when loading the cheats ) other than that works fine for me with my games thanks again for the replay and support

JoseAaronLopezGarcia, 2023-09-08:
Yeah that's how cwcheat works on pops, not really an ARK issue, it's the same on all CFW (except for Adrenaline), though it might be possible to do something about it.

krazynez, 2023-09-09:
> > Fixed in latest release.
> 
> Yeah I tried the New Build since I'm migrating my 3000 and the Cheatpops loads fine minor nitpick (regarding the background showing the game when loading the cheats ) other than that works fine for me with my games thanks again for the replay and support

patch for background now in latest pre-release.