PSP-Archive/ARK-4 #231: Unable to find the NoDRM Engine setting

Goldtuga, 2023-09-12:
### New Feature

Where can I find the NoDRM Engine setting to be able to run DLC? Also, even if finding it and enabling it, will one have to manually download and set a plugin and/or use ISO Tool to run some DLC, like LittleBigPlanet's?

JoseAaronLopezGarcia, 2023-09-13:
The NoDRM engine is NOT for DLC loading. For that you need `nploader` plugin.
There is no setting to disabled NoDRM engine, it's always active.

closed 2023-09-13 by JoseAaronLopezGarcia

Goldtuga, 2023-09-15:
But it was in Pro-CFW ans there was DLC that needed the NoDRM engine to be disabled in order to work.

Also, how can I make ARK read PSP DLC in the form of save data and specific DLC like LittleBigPlanet's? The games don't even detect the save data even though they share the same region.

JoseAaronLopezGarcia, 2023-09-15:
ARK doesn't let you disable the NoDRM engine as there is no real reason to do this. On PRO it could clash with other plugins, this is not an issue with ARK.

Goldtuga, 2023-09-15:
> ARK doesn't let you disable the NoDRM engine as there is no real reason to do this. On PRO it could clash with other plugins, this is not an issue with ARK.

How can I make ARK read PSP DLC in the form of save data and specific DLC like LittleBigPlanet's? The games don't even detect the save data even though they share the same region.