PSP-Archive/ARK-4 #232: ARK not reasging DLC

Goldtuga, 2023-09-15:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.64 Rev 4

### Steps to reproduce

How can I make ARK read PSP DLC in the form of save data and specific DLC like LittleBigPlanet's? The games don't even detect the save data even though they share the same region.

Also, in Pro-CFW there was DLC that needed the NoDRM engine to be disabled in order to work but in ARK NoDRM is always active.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

closed 2023-09-15 by krazynez

Goldtuga, 2023-09-15:
> LBP doesn't use savedata for DLC it uses it where GAME is located. I have LBP & DLC working fine on a Vita and the PSP with the same Savedata files.
> 
> use `nploader` to load DLC for LBP.

I wasn't referring to LBP when I was talking about DLC in the form of save data. I already use nploader and it's not working for LBP AND games whose DLC is available in the form of save data since the games don't even detect the save data even though they share the same region. According to what I searched, one mjst use a tool called ISO Tool to update LBP's ISO for it to load DLC.

JoseAaronLopezGarcia, 2023-09-15:
So wait, is this issue ONLY related to LBP or is it a generic issue? The tittle of this thread makes it seem like it's a generic issue with all DLC, which is simply not true, hence why the issue was closed. But if it's only an issue with LBP and we can confirm it then we might re-open it.

krazynez, 2023-09-15:
> So wait, is this issue ONLY related to LBP or is it a generic issue? The tittle of this thread makes it seem like it's a generic issue with all DLC, which is simply not true, hence why the issue was closed. But if it's only an issue with LBP and we can confirm it then we might re-open it.

I have already confirmed it works fine on both PSP and Vita. If you have LBP with the update 2.05 and DLC the DLC does show up fine. If you do not have the update it will not display. So yes @Goldtuga if your LBP is not updated update it with ISO Tool it will detect the PBOOT.PBP and update the EBOOT.BIN to 2.05.

krazynez, 2023-09-15:
Just also tested Wipeout Pure with DLC (located in Savedata) also working fine.

Goldtuga, 2023-09-15:
> So wait, is this issue ONLY related to LBP or is it a generic issue? The tittle of this thread makes it seem like it's a generic issue with all DLC, which is simply not true, hence why the issue was closed. But if it's only an issue with LBP and we can confirm it then we might re-open it.

It's a generic issue regarding DLC in save data format since the games don't detect the save data even though they share the same region and regarding specific games like LPB.

> I have already confirmed it works fine on both PSP and Vita. If you have LBP with the update 2.05 and DLC the DLC does show up fine. If you do not have the update it will not display. So yes [...]

How can I know if it has the 2.05 update or not? I assume it doesn't. Anyway, shouldn't it be mencioned in ARK's GitHub some DLC will only work after updating the ISO? Is using ISO Tool the only way to uodate ISOs?

And like I stated, DLC in the form of save data isn't recognized by the games since they don't appear when trying to load them even though they share the same region as their games.

JoseAaronLopezGarcia, 2023-09-15:
> 
> and it's another issue regarding specific games like LPB which need to be updated by ISO Tools for the DLC to be loaded.

If it's another issue then it should have it's own thread, but as @krazynez pointed out, this isn't an issue with ARK, it's an issue of you trying to load DLC on a version of the game that does NOT support DLC.

Nothing about this thread is related to ARK in any way, shape or form and it is not ARK's business to keep track of every single PSP game update. There's a whole internet out there that you can use to find information, use it.

In any case, this thread is solved so we're no longer going to reply to it.

Goldtuga, 2023-09-15:
> > It's a generic issue regarding DLC in save data format since the games don't detect the save data even though they share the same region as their games
> 
> This is simply NOT true. Nobody is having issues loading DLC with ARK and you haven't even provided a solid example of this happening.
> 
> > and it's another issue regarding specific games like LPB which need to be updated by ISO Tools for the DLC to be loaded.
> 
> If it's another issue then it should have it's own thread, but as @krazynez pointed out, this isn't an issue with ARK, it's an issue of you trying to load DLC on a version of the game that does NOT support DLC.
> 
> Nothing about this thread is related to ARK in any way, shape or form and it is not ARK's business to keep track of every single PSP game update. There's a whole internet out there that you can use to find information, use it.
> 
> In any case, this thread is solved so we're no longer going to reply to it.

It says no save data wasn't found when trying to load one (which contains DLC).

I already talked about both problems in the same issue and it shouldn't have been closed. How come it doesn't support DLC if it was released as save data and its egion matches the game's? Yes, there are other users having problems with loading DLC like LBP's. That's precisely how I found about the ISO Tool.

JoseAaronLopezGarcia, 2023-09-15:
You need to update the game.

Goldtuga, 2023-09-15:
> You need to update the game.

I know but that should be said in ARK's GitHub. What about the save data which contain DLC not being recognized by the games?

JoseAaronLopezGarcia, 2023-09-15:
Why should it be said in ARK's github? This has nothing to do with ARK. You'll get the same issue if your PSP is running official firmware.

Goldtuga, 2023-09-15:
> Why should it be said in ARK's github? This has nothing to do with ARK. You'll get the same issue if your PSP is running official firmware.

Because it mentions plugins, which are also used by other CFW. What about the DLC save data issue?