PSP-Archive/ARK-4 #233: Doubt about NoPspEmuDrm plugin

Goldtuga, 2023-09-16:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.64 Rev 4

### Steps to reproduce

When iARK's GitHub says to download and install the NoPspEmuDrm plugin and that one can use AutoPlugin II to do it, is it referring to AutoPlugin II's NoNpDrm Unofficial or to AutoPlugin II's npdrm libre (this one in the Adrenaline plugins section)? If to one of these two, may one remove (or not even install) the other one?

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

krazynez, 2023-09-16:
Good point it is this one: `NoPspEmuDrm by SilicaAndPina`

also changed it via Readme and Wiki as well to state which one.

closed 2023-09-16 by krazynez

Goldtuga, 2023-09-16:
> Good point it is this one: `NoPspEmuDrm by SilicaAndPina`
> 
> also changed it via Readme and Wiki as well to state which one.

But that plugin isn't available in AutoPlugin II so should one download it from the link mentioned in ARK's GitHub?

krazynez, 2023-09-16:
[image]
 Yes it is.

Goldtuga, 2023-09-16:
> [image] Yes it is.

You're right. Should I install it and keep NoNpDrm Unofficial or may I remove the latter? Do they do the same thing? Also, I found out there are now two plugins in AutoPlugin II's PSP section: Npdrm free by qwikrazor87 and Npdrm free by qwikrazor87, modified by lusid1. Are they both needed or just one of them?

krazynez, 2023-09-16:
That is the only one that is Officially supported.

Goldtuga, 2023-09-16:
> That is the only one that is Officially supported.

What do you mean? 

Anyway, I've been installing ARK on the PS Vita via the FasterARK method without installing the NoPspEmuDrm plugin and it has been working.

krazynez, 2023-09-16:
> > That is the only one that is Officially supported.
> 
> What do you mean?
> 
> Anyway, I've been installing ARK on the PS Vita via the FasterARK method without installing the NoPspEmuDrm plugin and it has been working.

The one we say in the README & Wiki is the only one that is officially supported for ARK-4.

Goldtuga, 2023-09-16:
> The one we say in the README & Wiki is the only one that is officially supported for ARK-4.

So, does it do a different thing than the NoNpDrm Unnoficial plugin, then?

Goldtuga, 2023-09-16:
> The one we say in the README & Wiki is the only one that is officially supported for ARK-4.

What's the difference between AutoPlugin II's PSP Npdrm free by qwikrazor87 and Npdrm free by qwikrazor87, modified by lusid1 plugins?

krazynez, 2023-09-16:
I do not know about other plugins, just the ones we support.

Goldtuga, 2023-09-16:
> I do not know about other plugins, just the ones we support.

I gues the latter needs nploader.