PSP-Archive/ARK-4 #234: PSP-1000 CXMB CTF Themes Not Showing Up

xexvisualz, 2023-09-16:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.64 r3

### Plugins ( if applicable )

vsh, ms0:/seplugins/cxmb.prx, on

### Steps to reproduce

1. move the cxmb.prx plugin to the "seplugins" folder
2. add "vsh, ms0:/seplugins/cxmb.prx, on" to PLUGINS.TXT
3. Confirm the CXMB plugin is enabled in Plugins Manager
4. Attempt to apply CTF themes (None shows up for me)

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-09-16:
Do you actually have themes installed in /PSP/THEME/?

xexvisualz, 2023-09-16:
> Do you actually have themes installed in /PSP/THEME/?

Yes, I have 5 CTF themes in the /PSP/THEME/ folder.

FULANO93, 2023-09-16:
I believe that you're not using the universal version of the plugin, as the universal version is inside a folder, inside the Seplugins folder, maybe that could be why it is not working, path error... I remember that in the version of this plugin for 6.20 it used a folder outside the Sepluigins folder

xexvisualz, 2023-09-17:
> I believe that you're not using the universal version of the plugin, as the universal version is inside a folder, inside the Seplugins folder, maybe that could be why it is not working, path error... I remember that in the version of this plugin for 6.20 it used a folder outside the Sepluigins folder and needs to be the first line in the VSH.TXT, now PLUGINS.TXT

I am using the universal version of the plugin. I also tried using the path "/PSP/SEPLUGINS/cxmb/cxmb.prx" and of course updating the path in the PLUGINS.TXT file to "vsh, ms0:/seplugins/cxmb/cxmb.prx, on" and still nothing unfortunately. Thank you for the suggestion

FULANO93, 2023-09-17:
Sometimes, for me, the plugin caused a bug where it didn't update the text file it creates with the name of the CTF Theme... I had to go there and change it manually and then the theme worked, this happened to me a few times using the plugin on firmware 6.20... With ARK-4 I just had the problem of having to install the CTF Theme several times for it to be really active and functional... an example of what I'm talking about was with the LittleBigPlanet theme... If you continue to have this problem, you can [download](https://www.mediafire.com/file/7hd610l9y6c1g01/HOMEPLUGINS.rar/file)  my seplugins folder with the plugins already installed and tested, just needing activation in the plugins menu

krazynez, 2023-09-17:
> Sometimes, for me, the plugin caused a bug where it didn't update the text file it creates with the name of the CTF Theme... I had to go there and change it manually and then the theme worked, this happened to me a few times using the plugin on firmware 6.20... With ARK-4 I just had the problem of having to install the CTF Theme several times for it to be really active and functional... an example of what I'm talking about was with the LittleBigPlanet theme... If you continue to have this problem, you can [download](https://www.mediafire.com/file/7hd610l9y6c1g01/HOMEPLUGINS.rar/file) my seplugins folder with the plugins already installed and tested, just needing activation in the plugins menu... my custom home plugin use the portuguese language...

Wow thanks for helping out, I did try the LBP theme out myself and worked fine on my 1K but if that does help out @xexvisualz thanks, we would still need to get to the bottom of the issue however. So if there is anything noticable @xexvisualz that differs let us know for sure!

JoseAaronLopezGarcia, 2023-09-17:
We should probably create a package with pre-installed and pre-configured plugins for ARK so users just have to extract the seplugins folder from such package.

krazynez, 2023-09-17:
> We should probably create a package with pre-installed and pre-configured plugins for ARK so users just have to extract the seplugins folder from such package.

huh you know that is a good idea. I guess I am being task that :joy:

xexvisualz, 2023-09-18:
> Sometimes, for me, the plugin caused a bug where it didn't update the text file it creates with the name of the CTF Theme... I had to go there and change it manually and then the theme worked, this [...]

I tried using your cxmb folder with the said Little Big Planet theme and I still unfortunately haven't solved it yet. I tried putting my PLUGINS.TXT in SEPLUGINS and in the /PSP/SAVEDATA/ARK_01234 folder and still nothing. Its almost like the plugin isn't enabled at all although it shows it is in the PLUGINS.TXT and in the recovery menu. Ill try to install another plugin and see if it does the same

JoseAaronLopezGarcia, 2023-09-18:
Try installing the PLUGINS.TXT in seplugins folder, I don't anybody has ever actually tested having their plugins.txt installed in ARK_01234.

xexvisualz, 2023-09-18:
> Try installing the PLUGINS.TXT in seplugins folder, I don't think anybody has ever actually tested having their plugins.txt installed in ARK_01234.

I tried that before with no luck unfortunately. However, i figured out what I did wrong. I realized that I accidentally made my SEPLUGINS folder in /PSP/ and not on the root of the memory stick. I also moved my PLUGINS.TXT in the SEPLUGINS folder as well. Thank you  guys so much for your help, i really appreciate it  and i apologize for my dumb mistake

JoseAaronLopezGarcia, 2023-09-18:
Glad we could be of help.

closed 2023-09-18 by JoseAaronLopezGarcia