PSP-Archive/ARK-4 #235: Permanent CFW step must be followed before the Full Flash Installation one or the PSP will boot into OFW after being restarted

Goldtuga, 2023-09-16:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.64 Rev 4

### Steps to reproduce

After all one must follow the Permanent CFW step before the Full Flash Installation one unlike what ARK's GitHub's Wiki states (that ARK_Full_Installer may be ran within live ARK) or after restarting the PSP it will boot into OFW. Could this be better explained in ARK's GitHub's README.md and ARK's GitHub's Wiki?

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-09-16:
"My CFW isn't permanent until I make it permanent" <- this is what you're complaining about.

How is this an issue again?

closed 2023-09-16 by JoseAaronLopezGarcia

Goldtuga, 2023-09-16:
> "My CFW isn't permanent until I make it permanent" <- this is what you're complaining about.
> 
> How is this an issue again?

Because you previously mentioned (as well as GitHub's Wiki does) the full flash installation step could be done from live ARK, so without following one of the two permanent CFW steps (Infinity or cIPL).