PSP-Archive/ARK-4 #236: Option to remove/revert ARK Full Flash

ReduxFlakes, 2023-09-16:
### New Feature

I have installed the ARK Full Flash by running the .ebp of ARK Full Installer but I don't like it and so I want to go back to only having ARK with Infinity but I don't know how to uninstall/revert back.
I have a PSP 3000 with ARK 4.20.64 Infinity

krazynez, 2023-09-17:
You can use ChronoSwitch to revert back to OFW then reinstall ARK with Infinity. Or just put the ARK_01234 back in SAVADATA and it will still act just like default ARK but if the memory stick is removed you will still have ARK's Full Installation.

krazynez, 2023-09-17:
Added in latest unstable release: https://github.com/PSP-Archive/ARK-4/releases/tag/v1694990290

Hold R when starting the Installer will remove files if found. Else it will still just install if no files are found or no buttons are pressed on boot.

closed 2023-09-17 by krazynez

ReduxFlakes, 2023-09-17:
Thanks, I runned cromoswitch and it's all good now!

krazynez, 2023-09-17:
> Thanks, I runned cromoswitch and it's all good now!

Yeah probably best solution, but having a "hidden" uninstaller does not hurt either for the full installer I guess.

JoseAaronLopezGarcia, 2023-09-17:
Just make the Full Installer work like cIPL installer, where it has a button prompt for install/unsintall.

krazynez, 2023-09-17:
added