PSP-Archive/ARK-4 #237: ARK often freezes the PS TV or crashes with the C1-2858-3 kernel error

Goldtuga, 2023-09-17:
### PSP/Vita Model

VTE-1000 (Vita TV)

### ARK Version

4.20.64 Rev 4

### Steps to reproduce

I Installed ARK via the FasterARK method and ARK often freezes the PS TV or crashes with the C1-2858-3 kernel error. It seems what causes this is enabling the Memory Speedup setting (I set it to "Always") but what's weird is that even after disabling this setting, these issues still happen, so one must never enable this setting after installing ARK.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-09-17:
I'm gonna need a little more information. Is it only happening on Vita TV? Do you have NoPspEmuDrm installed? What plugins are you using?

Goldtuga, 2023-09-17:
> I'm gonna need a little more information. Is it only happening on Vita TV? Do you have NoPspEmuDrm installed? What plugins are you using?

I didn't test it in the PS Vita. Yes, I have. This was my previous config before installing the NoPspEmuDrm plugin:

# ITLS
*KERNEL
ur0:tai/storagemgr.skprx
ur0:tai/0syscall6.skprx
ur0:tai/nonpdrm_un.skprx
ur0:tai/itls.skprx
ux0:app/PSPEMUCFW/sce_module/adrenaline_kernel.skprx
ur0:tai/nosleep.skprx
ur0:tai/reAuth.skprx
ur0:tai/staybright.skprx
# This file is used as an alternative if ux0:tai/config.txt is not found.
# For users plugins, you must refresh taiHEN from HENkaku Settings for
# changes to take place.
# For kernel plugins, you must reboot for changes to take place.
# henkaku.skprx is hard-coded to load and is not listed here
*main
# main is a special titleid for SceShell
ur0:tai/henkaku.suprx
*NPXS10015
# this is for modifying the version string
ur0:tai/henkaku.suprx
*NPXS10016
# this is for modifying the version string in settings widget
ur0:tai/henkaku.suprx

Also, how can I make Adrenaline load DLC? I already downloaded and set nploader and nploader_mod and it srill doesn't load them.

krazynez, 2023-09-17:
You have a few plugins that are not required anymore. I would suggest you do some research on plugins more. This does not seem to be and ARK issue. Having a lean config.txt works fine and properly setup config.txt for henkaku.

Goldtuga, 2023-09-17:
> You have a few plugins that are not required anymore. I would suggest you do some research on plugins more. This does not seem to be and ARK issue. Having a lean config.txt works fine and properly setup config.txt for henkaku.

Why are they not required, and if so, why are they still available? Anyway, I removed all I could with AutoPlugin II and ARK still freezes the PS TV (and the PS Vita) and crashes with the C1-2858-3 error if the Memory Speedup setting is set to "Always" and I'm forced to take the PS TV's power cable.

krazynez, 2023-09-17:
rebuild your database

Goldtuga, 2023-09-17:
> rebuild your database

Already did.

Goldtuga, 2023-09-18:
> rebuild your database

Also, it seems one doesn't have to set the nploader plugin in the standalone ARK (unlike in PSP's ARK) since it successfully loads PSP's DLC.

Goldtuga, 2023-09-18:
> rebuild your database

Also, it seems one doesn't have to set the nploader plugin in the standalone ARK (unlike in PSP's ARK) since it successfully loads PSP's DLC. Why is that?

RandomGuy2023, 2023-09-18:
> rebuild your database

I reinstalled the standalone ARK via the FasterARK method and ARK isn't freezing the PS TV or crashing with the C1-2858-3 kernel error.

closed 2023-09-18 by krazynez

Goldtuga, 2023-09-19:
After all, after installing ARK via the new FasterARK method, it's still freezing the PS TV and ARK is crashing with the C1-2858-3 kernel error after running a PSP ISO.