PSP-Archive/ARK-4 #238: ARK often freezes the PS TV and crashes with the C1-2858-3 kernel error

Goldtuga, 2023-09-19:
### PSP/Vita Model

VTE-1000 (Vita TV)

### ARK Version

4.20.64 Rev 4

### Steps to reproduce

After all, after installing ARK via the new FasterARK method, it's still freezing the PS TV and ARK is still crashing with the C1-2858-3 kernel error.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-09-19:
You probably have some corrupted or badly installed file somewhere. Nobody else is having any of these issues but you.
You also provide absolutely no information about the crash, how it happens, what happens before, what you do, etc.
I'll close this until you can provide a proper report and not this generic one.

Goldtuga, 2023-09-20:
> You probably have some corrupted or badly installed file somewhere. Nobody else is having any of these issues but you. You also provide absolutely no information about the crash, how it happens, [...]

I even reinstalled HENkaku. Yes, there are other users complaining about this issue in GBATemp.

I just run a PSP ISO and it crashes with a black screen. It doesn't respond to any command until it crashes with the said error.

There are other times I run ARK and it freezes even before reaching the part to choose a game and since it doesn't respond to any command, I have to take the PS TV's power cable. Other apps besides ARK are randomly freezing.