PSP-Archive/ARK-4 #239: ARK often freezes the PS TV and the PS Vita in the "ChovySign" part and crashes with the C1-2858-3 kernel error after running a PSP ISO

Goldtuga, 2023-10-01:
### PSP/Vita Model

PCH-2000 (Vita Slim), VTE-1000 (Vita TV)

### ARK Version

4.20.64 Rev 4

### Plugins ( if applicable )

After installing ARK via the FasterARK method and running it on the PS Vita and the PS TV, it often freezes even before running PSP games (like in the "ChovySign" part) and when running PSP ISOs with ARK, it often has a black screen for some seconds until it crashes with the C1-2858-3 kernel error.

### Steps to reproduce

Run ARK.
Run ARK and a PSP ISO.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-10-02:
Again with this shit? I'm tired of seeing this bug report with absolutely no information to reproduce. I will continue to close these as long as you continue to make them so increadibly useless.

closed 2023-10-02 by JoseAaronLopezGarcia

Goldtuga, 2023-10-02:
> Again with this shit? I'm tired of seeing this bug report with absolutely no information to reproduce. I will continue to close these as long as you continue to make them so increadibly useless.
> 
> You're not even telling me the firmware version of your Vita, how can I take you seriously?

That's my line. You're so tired of seeing it but you don't even know I'm on HENkaku 3.60 like I've told many times in other issues. Previously, you're excuse was that I wasn't describing the problem properly and now that I didn't mention the FW. Absolutely no information? Even if I creat another issue, you'll just find another excuse. 

Again with this shit? You don't fix the problem and then it's my fault? It's me who shouldn't take you seriously.

krazynez, 2023-10-02:
Yeah because you making issues and having all the threads edited the same time is not suspicious at all....

Goldtuga, 2023-10-02:
> Yeah because you making issues and having all the threads edited the same time is not suspicious at all....

Why so you say that? I do that on purpose just to make myself clear since I'm always being demanded more details and because some threads are closed against my will.