PSP-Archive/ARK-4 #241: ARK often freezes the PS Vita and the PS TV in the "ChovySign" part and crashes with the C1-2858-3 kernel error after running a PSP ISO

Goldtuga, 2023-10-02:
### PSP/Vita Model

PCH-2000 (Vita Slim), VTE-1000 (Vita TV)

### ARK Version

4.20.64 Rev 4

### Steps to reproduce

After installing ARK via the FasterARK method and running it on the PS Vita and the PS TV, it often freezes even before running PSP games (like in the "ChovySign" part) and when running PSP ISOs with ARK, it often has a black screen for some seconds until it crashes with the C1-2858-3 kernel error.

I'm on HENkaku 3.60.

[image]

[image]

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-10-08:
Badly copied K.BIN file.

Goldtuga, 2023-10-08:
> Badly copied K.BIN file.

What do you mean? I uploaded it via FTP and copied it via USB many times and I always had this issue since the beginning.

JoseAaronLopezGarcia, 2023-10-08:
The problem here is that nobody else is reporting this issue and nobody else can reproduce it.
We've had the CFW tested even on DevKit PS Vita units and it works perfectly fine.
Something must be wrong with your PS Vita. i.e. Wrong plugin, wrong install files, wrong version, etc.

closed 2023-10-08 by JoseAaronLopezGarcia