PSP-Archive/ARK-4 #242: Disabling UMD drive from recovery also disable analog stick

jappoman, 2023-10-03:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.64 Rev 4

### Steps to reproduce

1. Go to recovery
2. Disable umd drive only
3. Analog stick will be disabled too
4. Go to recovery again to re-enable the analog stick and exit
5. The analog is still disabled

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

krazynez, 2023-10-03:
Wow indeed it is. Good find on that one will fix very soon. Thanks for the report.

jappoman, 2023-10-03:
Also, just a question: disabling the umd drive can improve the battery life if i just play backups from memory stick?

krazynez, 2023-10-03:
Only if you have a UMD in at the same time. If there is no UMD then no its not going to make a difference.

jappoman, 2023-10-03:
Ok thank you!

krazynez, 2023-10-03:
Fixed in latest pre-release: https://github.com/PSP-Archive/ARK-4/releases/download/v1696331476/ARK-4-v1696331476.zip

JoseAaronLopezGarcia, 2023-10-03:
> Only if you have a UMD in at the same time. If there is no UMD then no its not going to make a difference.

It's the other way around, you need to have an empty UMD drive so it doesn't constantly try to read the disk.
This option doesn't electrically disable the UMD, it only makes the software think there isn't any, but if you have a disk in the tray it'll still spin and the laser will still shine, consuming battery.

krazynez, 2023-10-03:
> It's the other way around, you need to have an empty UMD drive so it doesn't constantly try to read the disk. This option doesn't electrically disable the UMD, it only makes the software think there [...]

Ah yes my mistake, was testing on a 2K think it has 2 switches for UMD detection. 1K does still light up and try to seek via laser.

JoseAaronLopezGarcia, 2023-10-08:
Fixed: https://github.com/PSP-Archive/ARK-4/releases/tag/r42065

closed 2023-10-08 by JoseAaronLopezGarcia