PSP-Archive/ARK-4 #243: ISOs no longer loading on multiple PSPgos

Larry70455, 2023-10-04:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.64 r4

### Homebrew/GAME (if applicable)

Crisis core, tekken 6, yugioh tag force, dissidia, persona 2, persona 3

### Steps to reproduce

1.Fresh install 6.61 (mine were 6.60 before so i just upgraded using Sony pbp, no chronoswitch here)
2. Place ARK4 and Infinity (Eboot_go.pbp renamed to eboot.pbp and placed in update)into respective savedata and game folder
3.run infinity twice, then run Ark loader
4 load any iso, doesn't work. PS1 games are fine. Homebrew is fine.

ive tried changing regions, i have enabled and disabled all recovery settings, not using any plugins. will go back to a previous version until fix

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-10-04:
This was fixed in latest pre-release.

Larry70455, 2023-10-04:
thank you will try
last working version for me was 
[Version 4.20.63 Rev 2](https://github.com/PSP-Archive/ARK-4/releases/tag/r42063)

JoseAaronLopezGarcia, 2023-10-08:
Fixed: https://github.com/PSP-Archive/ARK-4/releases/tag/r42065

closed 2023-10-08 by JoseAaronLopezGarcia