PSP-Archive/ARK-4 #244: Xmbih Not Working

FULANO93, 2023-10-08:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.65

### Plugins ( if applicable )

vsh, ms0:/seplugins/cxmb/cxmb.prx, off
vsh, ms0:/seplugins/categorieslite/category_lite.prx, off
vsh, ms0:/seplugins/dayviewer.prx, off
vsh, ms0:/seplugins/xmbih.prx, off
all, ms0:/seplugins/bright3.prx, off
all, ms0:/seplugins/noumd.prx, off
all, ms0:/seplugins/npdrm_free.prx, off
game, ms0:/seplugins/cushome/cushome.prx, off
game,ms0:/seplugins/pspstates/states_menu_pict_mod.prx, off
game, ms0:/seplugins/nploader.prx, off
pops, ms0:/seplugins/popsloader/popsloader.prx, off
pops, ms0:/seplugins/popsloader/cdda_enabler.prx, off
all, ms0:/seplugins/macrofire.prx, off
all,ms0:/seplugins/prxshot_png_mod/prxshot_png_mod.prx, off
all, ms0:/seplugins/prxshot++.prx, off
all, ms0:/seplugins/prxshot.prx, off
all, ms0:/seplugins/ssssspng.prx, off
all, ms0:/seplugins/sssssjpg.prx, off
all, ms0:/seplugins/sssssbmp.prx, off
all, ms0:/CheatMaster/CheatMaster.prx, off
game, ms0:/seplugins/cwcheat/cwcheat.prx, off
pops, ms0:/seplugins/cwcheat/cwcheatpops.prx, off
game, ms0:/seplugins/TempAR/tempar.prx, off
pops, ms0:/seplugins/TempAR/tempar_lite.prx, off

### Steps to reproduce

Just active the xmbih.prx plugin on the plugins and this dont works anymore... I think is the old bug with the xmbctrl, in this release have some changes with this and I think is because it... in the old Ark this was working...

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

krazynez, 2023-10-08:
its disabled in your plugins.txt

FULANO93, 2023-10-08:
No, I just forgot to edit because I took the plugins.txt that I have saved on my cell phone to report the bug, on the PSP it is active, I also tested activating the options individually and it didn't work, before I used some global options to hide the internet and psn

JoseAaronLopezGarcia, 2023-10-08:
It's not XMB Control, you might be tempted to think it is because the original XMB Control clashes with XMBIH, but the one in ARK is made to work fine with it.

No the real issue here is that XMBIH is one of those few plugins that can't detect 6.61 and errors out on it. This is usually fixed with the plugin 660on661 by Davee, but since that plugin causes some other issues I had cooked code into ARK to fix these plugins.

Now here's the thing, I was testing out this code by disabling it and totally forgot to re-enable it (you can see the code is commented-out here: https://github.com/PSP-Archive/ARK-4/blob/ced583e4348b41f696240f11c888f18912267d1a/core/systemctrl/src/syspatch.c#L80).

Anyways, I'm releasing a revision with the fix.

closed 2023-10-08 by JoseAaronLopezGarcia