PSP-Archive/ARK-4 #246: Individual Plugins System

FULANO93, 2023-10-08:
### New Feature

Old individual plugin activation and deactivation system, to activate and deactivate a specific section whenever you want, example: deactivate xmb plugins, something like this image... the first tree options...
[image]

JoseAaronLopezGarcia, 2023-10-09:
You can achieve the same result temporarily by using Vanilla Mode (https://github.com/PSP-Archive/ARK-4#vanilla-mode).
I don't think it makes any sense to permanently disable plugins.

FULANO93, 2023-10-09:
With the option to desactive a section of plugins it can be easier to know which one is causing some type of conflict or crash, so there is no need to desactive one by one or all at once... the ark launcher on my psp 2004 has been stuck on load since the 4.20.64 (it starts charging and then freezes, when I put the switch to off, the screen goes black as if it were about to go to sleep, but nothing happens and the led remains on, I have to put the switch to off again and only then does it crashes and turns off completely without being the normal way), in update 4.20.65 it started working normally again, but in rev1 it stopped loading again, I wanted to see if it was due to some plugin but without having to deactivate and test one by one ... would be good for testing...

krazynez, 2023-10-09:
> With the option to desactive a section of plugins it can be easier to know which one is causing some type of conflict or crash, so there is no need to desactive one by one or all at once... the ark [...]

yeah vanilla does just that its just temp. So from a coldboot if you press & hold Start (or Select) it will disable all vsh/xmb plugins on boot. If you launch a game your game plugins will be still enabled, unless you press and hold Start or Select on booting the game. Same thing goes for all runlevels.

JoseAaronLopezGarcia, 2023-10-09:
You can do literally the same with Vanilla mode. Hold start when loading XMB and it'll disable all plugins on the XMB, hold start when loading Custom Launcher and it'll disable all plugins for the Custom Launcher, hold start when loading a game (PSP or PS1) or a homebrew and it'll disable all plugins for that homebrew.

What you're asking for is literally what Vanilla Mode is for.

FULANO93, 2023-10-09:
Yeah yeah, I've already tried the commands described in the read file, it keeps crashing, I just updated to rev2 and it keeps crashing, in the end it must be my problem after all... it was just a suggestion, which to me seemed good, have a good afternoon...

closed 2023-10-09 by FULANO93

JoseAaronLopezGarcia, 2023-10-09:
> Yeah yeah, I've already tried the commands described in the read file, it keeps crashing, I just updated to rev2 and it keeps crashing, in the end it must be my problem after all... it was just a suggestion, which to me seemed good, have a good afternoon...

Maybe you can open a new bug report. If there's a bug/regression in the custom launcher then I want to fix it.
You mentioned 4.20.64 being the first version that started freezing, is 4.20.63 working?

You can open a new bug report with as much information as you can gather.