PSP-Archive/ARK-4 #247: Option to change location of CFW settings and plugins manager in xmb

PSPGOENTHUSIAST, 2023-10-09:
### New Feature

Hello, sorry for the long title. I just started using this CFW in the last two weeks and just today I updated it and noticed that the plugins manager and CFW settings moved to the game section of the xmb. I'm just requesting there be an option to move them back to the settings section of the xmb once again. Thank you

JoseAaronLopezGarcia, 2023-10-09:
The reason we moved it is because it was causing issues with the icons in the settings section, nothing too big only visual, but some users didn't like it.

We would like to move it again, but over to the Extras section, which we're still investigating how to do, though the idea of making it configurable is not so bad, I generally dislike adding new settings to keep the CFW simple and easy to use.

Another option would be to add it to the end of the settings section instead of at the top, just exactly where Debug Settings and Package Manager is located on Testkits/Devkits.

It is certaintly something we'll continue to investigate to find a better placement.

JoseAaronLopezGarcia, 2023-10-09:
As of 4.20.65 Rev 2 the CFW settings and plugins manager will now be located in the Extras section. We believe this is the best place for it.
However if that section doesn't exist (i.e. the region), then they will be sent to the Games section instead.
You can use Advanced VSH Menu to change VSH region to one that has an extras section.

closed 2023-10-09 by JoseAaronLopezGarcia