PSP-Archive/ARK-4 #248: Custom Launcher Freeze

FULANO93, 2023-10-09:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.65 Rev3

### Plugins ( if applicable )

vsh, ms0:/seplugins/cxmb/cxmb.prx, on
vsh, ms0:/seplugins/categorieslite/category_lite.prx, on
vsh, ms0:/seplugins/dayviewer.prx, on
vsh, ms0:/seplugins/xmbih.prx, on
all, ms0:/seplugins/bright3.prx, on
all, ms0:/seplugins/noumd.prx, off
all, ms0:/seplugins/npdrm_free.prx, off
game, ms0:/seplugins/cushome/cushome.prx, on
game, ms0:/seplugins/pspstates/states_menu_pict_mod.prx, on
game, ms0:/seplugins/nploader.prx, on
pops, ms0:/seplugins/popsloader/popsloader.prx, on
pops, ms0:/seplugins/popsloader/cdda_enabler.prx, off
all, ms0:/seplugins/macrofire.prx, off
all, ms0:/seplugins/prxshot_png_mod/prxshot_png_mod.prx, off
all, ms0:/seplugins/prxshot++.prx, off
all, ms0:/seplugins/prxshot.prx, off
all, ms0:/seplugins/ssssspng.prx, off
all, ms0:/seplugins/sssssjpg.prx, off
all, ms0:/seplugins/sssssbmp.prx, off
all, ms0:/CheatMaster/CheatMaster.prx, off
game, ms0:/seplugins/cwcheat/cwcheat.prx, on
pops, ms0:/seplugins/cwcheat/cwcheatpops.prx, on
game, ms0:/seplugins/TempAR/tempar.prx, on
pops, ms0:/seplugins/TempAR/tempar_lite.prx, off

### Steps to reproduce

Just try open the custom launcher and it freeze...

4.20.63 Working
4.20.64 It started to freeze with one of the revisions, I don't remember exactly which one
4.20.65 Working
4.20.65 Rev 1,2,3 Freeze

[Vídeo](https://youtu.be/t4NCTfUbDyc?si=4Bin8nGV2K_Z3VmF) 

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-10-09:
How is 4.20.65 Working but the revs aren't? None of the revs are related to the launcher.

FULANO93, 2023-10-09:
Maybe because in 4.20.65 the 660 to 661 integrated plugin was not working, maybe it is causing conflict

JoseAaronLopezGarcia, 2023-10-09:
But didn't you test it with all plugins disabled and it still crashes?

FULANO93, 2023-10-09:
Yeah, I will test it again...

JoseAaronLopezGarcia, 2023-10-09:
Also you say that 4.20.63 is Working and that one has the 6.60 on 6.61 plugins fix.
It must be something else.

FULANO93, 2023-10-09:
In 4.20.64 I remember that in the first versions it also working

JoseAaronLopezGarcia, 2023-10-09:
Perhaps you can try pre-releases until you come accross one that works and one that doesn't.

FULANO93, 2023-10-09:
Updated to ARK-4-v1696867941 pre-release using the update folder and still freeze

krazynez, 2023-10-09:
Curious which language are you using for XMB ?

FULANO93, 2023-10-09:
My language is Portuguese

krazynez, 2023-10-09:
> ARK-4-v1696867941

Oh @JoseAaronLopezGarcia means try from 4.20.65 pre-releases until you hit rev1 or you hit the commit that freezes it.

FULANO93, 2023-10-09:
ARK-4-v1696721794 pre release works, at the time of installation it said 4.20.65 rev0, but in this version there was only the plugins option in the game tab...here for me this is the newest version before version 4.20.65 rev3 to download...either I'm doing it wrong or releases prior to rev3 don't appear for me...

[image]

krazynez, 2023-10-09:
They will if you scroll down and keep going back in the pages. Your on the right path though. That is **ALL** release so if pre-releases or "test" that we call it.

FULANO93, 2023-10-09:
It's like I said, there is no pre-releases after this ARK-4-v1696721794, this is the last before the current release 4 20.65 Rev3, so if you can find pre-releases before rev3 and higher than rev0, show me the way

krazynez, 2023-10-09:
> It's like I said, there is no pre-releases after this ARK-4-v1696721794, this is the last before the current release 4 20.65 Rev3, so if you can find pre-releases before rev3 and higher than rev0, show me the way...I'll see later, now I need to take my grandma on his walk

https://github.com/PSP-Archive/ARK-4/releases?page=2

FULANO93, 2023-10-09:
I just saw where I was going wrong, here it shows the pre-releases up to the current version, but after that they are still the ones before revision 1, I'll do some tests until I see which one starts to crash

FULANO93, 2023-10-11:
> > I just saw where I was going wrong, here it shows the pre-releases up to the current version, but after that they are still the ones before revision 1, I'll do some tests until I see which one starts to crash
> > Edit: v1696727905 Freeze v1696725422 Works
> 
> [v1696950397](https://github.com/PSP-Archive/ARK-4/releases/tag/v1696950397) does this one work?

Tested, still freeze on this version

krazynez, 2023-10-11:
> I just saw where I was going wrong, here it shows the pre-releases up to the current version, but after that they are still the ones before revision 1, I'll do some tests until I see which one starts to crash
> 
> Edit: v1696727905 Freeze v1696725422 Works

This is confusing because the only difference was blanking the screen was commit difference as seen [here](https://github.com/PSP-Archive/ARK-4/compare/v1696725422...v1696727905)

FULANO93, 2023-10-11:
If it's confusing for you, imagine for me that I don't understand anything about programming... what I can do is just test and report... in the revisions of the previous version to the current one, the launcher also started to crash, something that didn't happen in the previous version primary (rev0). Could this happen due to review? something corrupted on the card? Due to something wrong with my card, the launcher would freeze, but this happened after it started and tried to load the games, I solved it by formatting and passing all the games again... before I was using a adapter with two SD cards, one 64 and a 32... now I'm just using an original sony 32 card.

4.20.64 rev0 ok rev1 and 2 i dont remember, but rev3 4... freeze
4.20.65 rev0 ok rev1 2 3 4 5 freeze... 
My PSP is an 2004 2g

FULANO93, 2023-10-11:
My mistake, I forgot that I hadn't tested revision 5, I just tested it and the launcher is working on it

krazynez, 2023-10-11:
> My mistake, I forgot that I hadn't tested revision 5, I just tested it and the launcher is working on it

So we are good to close this then?

FULANO93, 2023-10-11:
Yes, I believe the problem was resolved in revision 5, but I'm a little unsure of what actually causes the freezing.

krazynez, 2023-10-11:
> Yes, I believe the problem was resolved in revision 5, but I'm a little unsure of what actually causes the freezing.

Yeah same here, but for now we can close it if it happens we can reopen the issue again. :+1:

closed 2023-10-11 by krazynez