PSP-Archive/ARK-4 #249: CSO Shutoff From Memory stick

PSPGOENTHUSIAST, 2023-10-10:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.65 r3

### Homebrew/GAME (if applicable)

All CSO Games from Memory Stick

### Plugins ( if applicable )

xmb, ef0:/seplugins/CXMB/CXMB.PRX, on

game, ef0:/seplugins/CWCHEAT/cwcheat.prx, on

ps1, ef0:/seplugins/CWCHEAT/cwcheatpops.prx, on

game, ef0:/seplugins/nploader.prx, off

ps1, ef0:/seplugins/cdda_enabler.prx, on

ULUS10160, ef0:/seplugins/CWCHEAT/cwcheat.prx, off

ULUS10160, ef0:/seplugins/cheatdevice_remastered.prx, on

always, ef0:/seplugins/brightness.prx, on

xmb, ef0:/seplugins/category_lite.prx, on

### Steps to reproduce

Attemping to launch any CSO formatted game stored on my memory stick from either the xmb or the custom launcher results in my PSP go shutting off.

Extra details:
Homebrew, iso files and PS1 eboots still launch and work from memory stick. The CSO games do work from memory stick when being ran under lme 2.3 CFW with the same plugin setup. CSO files were compressed using umdgen with compression level 9 selected. 

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Alway

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-10-10:
Unfortunately I can't seem to be able to reproduce this. Do you know the exact release that broke it?

PSPGOENTHUSIAST, 2023-10-10:
It hasn't ever worked for the past few days and even after a few updates still nothing. One thing I forgot to mention is that I am using an internal 128gb SD to memory stick adapter. If all else fails, I could get them all as iso files

JoseAaronLopezGarcia, 2023-10-10:
It only affects CSO? ISO works fine?

PSPGOENTHUSIAST, 2023-10-10:
Yes only CSO. Although the ISO loads slow but that's probably due to the 128gb size

PSPGOENTHUSIAST, 2023-10-10:
It was not formatted with the psp itself. It shouldn't be fragmented as it's a new card. I've read only that the only issue is longer xmb load times. I sure hope it hasn't gone fairly since it's only about a year old ordered from Etsy. 

I'll format it using the PSP and re copy everything and see what happens. Any way to make the xmb or custom launcher cache the games to reduce load times?

JoseAaronLopezGarcia, 2023-10-10:
The XMB already uses a cache for ISO files.

PSPGOENTHUSIAST, 2023-10-10:
After formatting using the PSP I still can't get CSO files to work. I'll just convert them to iso

JoseAaronLopezGarcia, 2023-10-10:
Try using ZSO or JSO, they are very fast.

PSPGOENTHUSIAST, 2023-10-10:
What windows programs can create those?

krazynez, 2023-10-10:
https://github.com/PSP-Archive/ARK-4/tree/main/contrib/PC/ciso

PSPGOENTHUSIAST, 2023-10-10:
Thanks. Any difference between zso or jso?

JoseAaronLopezGarcia, 2023-10-10:
- Maxcso can create CSO, ZSO and DAX files of any block size but it's only command-line.
- PSP ISO Compressor v1.4 supports CSO, JSO and DAX with a nice GUI but doesn't allow changing block size.
- PSXiSO is a recently made GUI app that can compress ISO to CHD and ZSO.

As for JSO vs ZSO vs CSO vs DAX:
- ZSO is the fastest but has the worst compression.
- CSO and DAX are the slowest but have better compression.
- JSO is a middle ground, better speed than CSO and better compression than ZSO.

PSPGOENTHUSIAST, 2023-10-10:
Perfect thanks I'll try them

JoseAaronLopezGarcia, 2023-10-10:
If speed really is a must then you have to go for ZSO with 8K blocks, even if compression isn't as good as other formats (it's still better/smaller than ISO in most cases, while maintaining ISO speeds).

The only app that lets you create these is maxcso, which is a command-line tool.

PSPGOENTHUSIAST, 2023-10-10:
Can you give me the command to create a zso with 8k blocks using Maxcso? I'm not the best with command line

JoseAaronLopezGarcia, 2023-10-10:
it should be something like this:

maxcso --block=8192 --format=zso "My Game.iso"

PSPGOENTHUSIAST, 2023-10-10:
I see. I just figured it out with jiso.exe. Should I still use Maxcso instead?

JoseAaronLopezGarcia, 2023-10-10:
Try both and stick to whichever works best for you.

PSPGOENTHUSIAST, 2023-10-10:
Is the compression algorithm for zso on jiso lzo or zlib?

JoseAaronLopezGarcia, 2023-10-10:
jiso is for JSO not for ZSO

PSPGOENTHUSIAST, 2023-10-10:
Gotcha. Jso files do boot from memory card

PSPGOENTHUSIAST, 2023-10-10:
Unrelated but do y'all have a discord so I don't flood your GitHub page with issues or ideas?

krazynez, 2023-10-10:
> Unrelated but do y'all have a discord so I don't flood your GitHub page with issues or ideas?

Appreciate that: We are apart of the [PSP Homebrew Community](https://discord.gg/bePrj9W) discord

JoseAaronLopezGarcia, 2023-10-10:
> > Unrelated but do y'all have a discord so I don't flood your GitHub page with issues or ideas?
> 
> Appreciate that: We are apart of the [PSP Homebrew Community](https://discord.gg/bePrj9W) discord

We should put this link somewhere easy to find.

PSPGOENTHUSIAST, 2023-10-10:
Technically I still can't use CSO files but zso are much better so we can consider this solved. Probably just an issue with my setup

krazynez, 2023-10-11:
verified OP stated can close via Discord

closed 2023-10-11 by krazynez