PSP-Archive/ARK-4 #258: More language fonts

Kitki30, 2023-10-15:
### New Feature

Can you make fonts for languages like polish so I can make Ł in translation not L

krazynez, 2023-10-15:
https://github.com/PSP-Archive/ARK-4/blob/main/extras/menus/arkMenu/themes/translations/resources/lang_pol.json

Please do update much appreciated

JoseAaronLopezGarcia, 2023-10-15:
I'm not sure, we're using Sony's official PGF format, and we know it's possible to convert TTF fonts to PGF fonts, but even with that we failed with the Greek language (we had to do a workaround).
I haven't tested with other languages, might be worth a try.

Kitki30, 2023-10-18:
@krazynez im gonna fix some translation errors

Kitki30, 2023-10-18:
> https://github.com/PSP-Archive/ARK-4/blob/main/extras/menus/arkMenu/themes/translations/resources/lang_pol.json
> 
> Please do update much appreciated 

Accept pull request please
I fixed polish language (eg. L instead of Ł that psp is not recognizing)
Notify me when there is more to translate in my language

JoseAaronLopezGarcia, 2023-11-02:
Closing this as I think it's been solved.

closed 2023-11-02 by JoseAaronLopezGarcia