PSP-Archive/ARK-4 #260: Any way of enable plugin PER homebrew?

pmsobrado, 2023-10-16:
### New Feature

I know I can enable a plugin for a certain game, with its ID. I can also enable a plugin for all homebrew. But can I enable a plugin only for a specific homebrew?

Thanks!

JoseAaronLopezGarcia, 2023-10-16:
We can't do this cause homebrew don't properly use Game ID. They all use the same ID, some of them use Game ID from real games, etc. There's no consistency in homebrew when it comes to the ID so we can't distinguish one from the other.

JoseAaronLopezGarcia, 2023-10-16:
That might be possible yes, via path. We actually do something similar with "launcher" keyword, we test the path to see if you're running the launcher.

Let me do something real quick and you can test it for me.

pmsobrado, 2023-10-16:
Cool :)

JoseAaronLopezGarcia, 2023-10-16:
You can try it here: https://github.com/PSP-Archive/ARK-4/releases/tag/v1697454709

You need to specify the full path like this:

ms0:/PSP/GAME/myhomebrew/EBOOT.PBP, ms0:/seplugins/myplugin.prx, on

or "off" if you need to disable the plugin on that specific homebrew.
Note that you can't have more than one path or combine path with regular keywords.

It also works for settings:

ms0:/PSP/GAME/myhomebrew/EBOOT.PBP, highmem, on

(or off to disable a setting on specific homebrew).

This will not work for ISO though, use GameID for those.

pmsobrado, 2023-10-16:
When you say "Note that you can't have more than one path or combine path with regular keywords", you mean:

ONE

This works:

ms0:/PSP/GAME/myhomebrew1/EBOOT.PBP, ms0:/seplugins/myplugin.prx, on
ms0:/PSP/GAME/myhomebrew2/EBOOT.PBP, ms0:/seplugins/myplugin.prx, on

And this does not:

ms0:/PSP/GAME/myhomebrew1/EBOOT.PBP, ms0:/PSP/GAME/myhomebrew2/EBOOT.PBP, ms0:/seplugins/myplugin.prx, on

Correct?

TWO

This kind of stuff won't work:

ms0:/PSP/GAME/myhomebrew*/EBOOT.PBP, ms0:/seplugins/myplugin.prx, on
ms0:/PSP/GAME/myhomebrew(1|2)/EBOOT.PBP, ms0:/seplugins/myplugin.prx, on
ms0:/PSP/GAME/myhomebrew/*/EBOOT.PBP, ms0:/seplugins/myplugin.prx, on
...

Correct?

BTW, I guess this is totally compatible with Game Categories Lite, right? I mean, because of the slightly different paths.

JoseAaronLopezGarcia, 2023-10-16:
Yes, yes and yes.

pmsobrado, 2023-10-16:
Great, I'll try and post back.

pmsobrado, 2023-10-17:
It work fine! I will keep an eye for official update. BTW, how to update standalone ARK on a Vita, just run FasterARK.vpk again?

krazynez, 2023-10-17:
> It works fine! I will keep an eye for official update. BTW, how to update standalone ARK on a Vita, just run FasterARK.vpk again?

Awesome! You update it in the Network tab.

pmsobrado, 2023-10-17:
You're right xD I'm a bit confused because I have Adrenaline, ARK through Adrenaline and standalone ARK (I only have standalone because of the Family Guy EUR versions that lags on Adrenaline/ARK Adrenaline for some reason).

pmsobrado, 2023-10-26:
@JoseAaronLopezGarcia Hi, is this feature available on latest Version 4.20.66 Rev 2 release? Thanks!

closed 2023-10-26 by pmsobrado

JoseAaronLopezGarcia, 2023-10-26:
Yes, it's available in latest release.

pmsobrado, 2023-10-26:
Great news, thanks!