PSP-Archive/ARK-4 #263: Custom launcher titles only for performance

jappoman, 2023-10-17:
### New Feature

As the title says, is it correct to assume that loading only the titles of games and homebrew will improve the startup time? If yes, can we have an option to enable this and skip entirely load of pic0 and pic1 ? 

Thank you in advance

JoseAaronLopezGarcia, 2023-10-17:
Pic0 and Pic1 are only loaded when you press X/O and it enters the game screen animation. The custom launcher already does the bare minimum when scanning games to make it as fast as possible, and trust me it is considerably faster than the XMB at scanning games.

The only thing that could make it faster at this point would be some sort of cache, which wouldn't be a bad idea to implement.

jappoman, 2023-10-18:
That could be really awesome. The custom launcher is already my default launcher and make it faster will only raise the bar for this awesome project. In the PSP scene I think that the only missing "feature" that a cfw can offer more than others is the speed. If you manage to achieve this, this will be another big selling point for this cfw.

JoseAaronLopezGarcia, 2025-03-16:
Startup time of Custom Launcher has been heavily improved since, so I'm closing this.

closed 2025-03-16 by JoseAaronLopezGarcia