PSP-Archive/ARK-4 #264: Fix pro online

Kitki30, 2023-10-18:
### New Feature

Pro online isnt loading
No menu on PS/Home button just normal psp
Do you want to close the game?
Is this ark error it showed normal on first install

JoseAaronLopezGarcia, 2023-10-18:
PRO Online works fine, how exactly did you install it?

Kitki30, 2023-10-18:
Get it from wiki transfer over ftp move with launcher to root
Maybe launcher changed name?
Can you disable this shit that shorts names of folders/files when copying

JoseAaronLopezGarcia, 2023-10-18:
It shouldn't short names unless they have special non-ascii characters in them, which PRO Online shouldn't have (the file is called atpro.prx).

Use this tutorial to install the plugin on specific games, only install the plugin on games known to work with PRO Online (there are compatibility lists online, search for them).
https://github.com/PSP-Archive/ARK-4/wiki/Plugins#per-game-plugin-install

krazynez, 2023-10-18:
I have tested it with Socom FTB and it was working fine. Which game are you trying to test with?

Kitki30, 2023-10-19:
> I have tested it with Socom FTB and it was working fine. Which game are you trying to test with?

Tekken 6
Tried both on UPnP and DNZ
Tekken is downloaded from random website

Kitki30, 2023-10-19:
On socom and sushicat servers

JoseAaronLopezGarcia, 2023-10-19:
Did you enable "Force Extra Memory"? It's needed for PRO Online.

Kitki30, 2023-10-19:
Yes

Kitki30, 2023-10-19:
Btw can you merge pull request i made satelite and that xmb file for pl translation

Kitki30, 2023-10-19:
[image]

[image]

[image]

krazynez, 2023-10-19:
I would only use atpro on which ever game specified and also Force Extra Memory its also best to just say UMD/ISO. Both having it set always or for any game can cause issues.

So I tested Tekken 6 out and it connects but I get no one in the lobby, however I tested on PPSSPP and I also had the same results.

Kitki30, 2023-10-19:
> I would only use atpro on which ever game specified and also Force Extra Memory its also best to just say UMD/ISO. Both having it set always or for any game can cause issues.
>
> So I tested Tekken 6 [...]

Yes no one is in lobby but you are also not shpwing on socom/sushicat list

JoseAaronLopezGarcia, 2023-10-19:
To be fair I too am having issues with PRO Online, but I know it worked before, the only change I made is the Wifi Hotspot, so it probably doesn't like the one I'm using right now.

PRO Online is known to be very finicky and is hard to get working.

JoseAaronLopezGarcia, 2023-11-02:
Closing as not ARK-related (PRO Online is a different, separate project).

closed 2023-11-02 by JoseAaronLopezGarcia