PSP-Archive/ARK-4 #268: Custom launcher exit game hotkey not working if using HUD plugin (very minor issue)

pmsobrado, 2023-10-19:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.65 Rev 5

### Homebrew/GAME (if applicable)

Any

### Plugins ( if applicable )

"game, ms0:/seplugins/hud.prx, on"

### Steps to reproduce

1. Launch game with HUD plugin enabled.
2. Try to exit to launcher.

It is a minor and unimportant bug, but I though I could ask here. This is the plugin: https://www.gamebrew.org/wiki/PSP_Hud

This only happens with the custom launcher (since the hotkey I'm trying is from and for the custom luncher, duh!). I guess the plugin captures all buttons first or something like that. It doesn't matter if configured keys on the plugin are not any of the ones of the hotkeys.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2023-10-19:
[image]

Custom Launcher shortcut doesn't use select?

pmsobrado, 2023-10-19:
You're using XMB, I mean, when in custom launcher, try to exit a game back to the gustom launcher with HUD plugin activated for game (doesn't need to be showed).

krazynez, 2023-10-19:
I am not using anything that is a screenshot of gamebrew 😂 I will test in a bit though.

pmsobrado, 2023-10-19:
Don't look at it too much, it's a silly bug :)

pmsobrado, 2023-10-19:
In any case, it doesn't matter if the plugin is using different keys/buttons. That's the weird part. The plugin must be capturing the keys somehow or whatever. Although, now that I think of it, I can use HUD and cwcheat at the same time...

krazynez, 2023-10-19:
Indeed it does how interesting.

JoseAaronLopezGarcia, 2023-10-19:
I got it.

pmsobrado, 2023-10-19:
You mean the cause?

krazynez, 2023-10-19:
> You mean the cause?

No he as a solution to fix it.

pmsobrado, 2023-10-19:
@JoseAaronLopezGarcia Can I use that update for standalone and Adrenaline versions of ARK? Will the "plugins per homebrew" build I'm using keep that change?

krazynez, 2023-10-19:
rev 5 has those fixes you should be fine.

pmsobrado, 2023-10-20:
@JoseAaronLopezGarcia I don't think it is on rev 5, that one is from two weeks ago and I'm talking about this:

https://github.com/PSP-Archive/ARK-4/issues/260

Thanks!

pmsobrado, 2023-10-26:
@JoseAaronLopezGarcia Hi, is this fix available on latest Version 4.20.66 Rev 2 release? Thanks!

JoseAaronLopezGarcia, 2023-10-26:
Yes, it should be fixed on latest release.

pmsobrado, 2023-10-26:
Great news, thanks!

pmsobrado, 2023-10-28:
I'm testing on latest 4.20.66 version and it is not fixed :(

pmsobrado, 2023-10-30:
@JoseAaronLopezGarcia should we reopen this issue for further testing?

krazynez, 2023-10-30:
I will yeah this also correlates with another recent issue as well it seems. https://github.com/PSP-Archive/ARK-4/issues/277

pmsobrado, 2023-10-30:
I don't know what CFE is, but, seems like there's a crash when exiting launcher, maybe is not directly related to this issue, but it's definitely affecting it.

pmsobrado, 2023-11-21:
Any news on this?

pmsobrado, 2024-02-04:
Is this fixed on latest release?

JoseAaronLopezGarcia, 2024-02-04:
No idea. Try it out and report back.

pmsobrado, 2024-02-04:
Yep, seems to be even more broken, I just opened another issue:

https://github.com/PSP-Archive/ARK-4/issues/301

Thanks!

JoseAaronLopezGarcia, 2024-02-04:
Looking into it.

pmsobrado, 2024-02-04:
@JoseAaronLopezGarcia any way to disable certain plugin only on standalone launcher?

JoseAaronLopezGarcia, 2024-02-04:
yes, you can use the "launcher" keyword for settings/plugins that affect only the launcher, letting you enable/disable only for it.
Simply write something like this anywhere after the line that enables the plugin:

launcher, ms0:/seplugins/myplugin.prx, off

I should probably add Launcher as an option when installing plugins, but for now I'm trying to figure out this one issue.

pmsobrado, 2024-02-04:
Cool, thanks.

JoseAaronLopezGarcia, 2024-02-04:
Got it working: https://github.com/PSP-Archive/ARK-4/releases/tag/v1707075722

This was a tough one, I had to slightly redo the way ARK applies the control patches so it was done first (easy), but then it kept crashing when trying to exit (this didn't happen with the HUD plugin disabled).
So I had to add some code to terminate the main thread of the HUD plugin before exiting, this fixes it and it's all good now.

Let me know if it works fine and all.

pmsobrado, 2024-02-04:
Trying it now...

JoseAaronLopezGarcia, 2024-02-04:
I just found a much better fix for it. The real cause of the issue was that HUD plugin creates a thread that polls the controller, and the game itself is also polling the controller, which made more than one thread attempt to exit and that caused a crash.
The new implementation doesn't patch the HUD plugin anymore, instead it implements detection of the exit thread already being executed, thus no more than one exit thread can run at the same time.
This fix should be much better as it should also fix any other test case where this might happen (i.e. any other plugin or homebrew).
You can find the new fix here: https://github.com/PSP-Archive/ARK-4/releases/tag/v1707077210

pmsobrado, 2024-02-04:
That one works, the other did not, so mission accomplish I guess! Thanks!

Will this fix be included on the next release and onwards?

JoseAaronLopezGarcia, 2024-02-04:
Rev 7 is now live as the main release.

As for those features:

1- HUD will do whatever patches, so it's normal that they conflict. This is a weird conflict indeed, but thankfully not that big of a deal.

2- ARK-4 used to have a recovery menu that was essentially a stripped down Custom Launcher, eventually we made the Classic Recovery Menu that is more akin to what you'd find in M33/PRO/ME/etc, to reduce that much redundancy we just merged the recovery menu into the custom launcher and kept the classic one as the standard. So bring back the original Recovery Menu you can now boot up the Custom Launcher, and hold the R button when you see the spinning wheel (launcher booting up).

pmsobrado, 2024-02-04:
I will download Rev 7 again, thanks.

As for the explanations, thanks very much. I guess that's the small menu I'm seeing, with those few options. Wait, isn't those options the same as "Normal boot -> CFW options"?

JoseAaronLopezGarcia, 2024-02-04:
Yeah, the Recovery Menu is basically there to handle CFW Settings, Plugins and File Browser. Basically the bare minimum needed to fix any issue.

pmsobrado, 2024-02-04:
Thanks for all the work!

closed 2024-02-04 by JoseAaronLopezGarcia