PSP-Archive/ARK-4 #276: Wont save iso drivers

damo08, 2023-10-28:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

Ark 4.20.66 infinity 

### Steps to reproduce

Press select
Click advanced vsh menu
Change iso driver to sony np9660
Go back
Reset vsh

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Homebrew

### Memory Stick Speedup

XMB

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2023-10-28:
https://github.com/PSP-Archive/ARK-4/issues/92#issuecomment-1450968108

damo08, 2023-10-28:
On pro cfw it saves the change

krazynez, 2023-10-28:
Its a reference to let the user know which driver works for which format. Hense the section to the right of the Driver name.

damo08, 2023-10-28:
Fixed flatout game by enabling inferno cache rr

closed 2023-10-28 by damo08