PSP-Archive/ARK-4 #280: Hang at boot when mounting ISO with non-latin characters

WindedHero, 2023-11-15:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.66 Infinity

### Steps to reproduce

- Place a UMD Video ISO into \ISO\VIDEO\ (can use プレイステーション.iso as an example)
- select the iso from the VSH mount
- system reboots but doesn't load the xmb
- hit select to pull up vsh and reset device
- rename ISO without non-latin characters, works fine

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2023-11-16:
Try using the Custom Launcher to mount these. Just use the File Browser to navigate to /ISO/VIDEO/ and open the ISO.
Let me know if that works.

WindedHero, 2023-11-18:
It took some work to figure it out, but the custom launcher does mount them properly. 
Navigating to the correct location is not simple through the custom launcher and the filenames are inaccurate (it seems to strip out all non-latin characters). 
Video ISO do not display on the main screen as a tile.

JoseAaronLopezGarcia, 2025-03-16:
Closing as fixed.

closed 2025-03-16 by JoseAaronLopezGarcia