PSP-Archive/ARK-4 #281: System crash while ARK-4 install

gr3ddy, 2023-11-18:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.66 Rev 3

### Steps to reproduce

1. Rollback firmware from 6.61 to 6.60
2. Copy ARL_01234 to `/PSP/SAVEDATA/` folder
3. Copy ARK_Loader to `/PSP/GAME/`. folder
4. Start ARK Loader

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2023-11-18:
[ARK4_Files.zip](https://github.com/PSP-Archive/ARK-4/files/13400985/ARK4_Files.zip)

Try using both of these.

gr3ddy, 2023-11-18:
This error was not on 6.61 and 4.20.66 Rev 2 (and normally updated to rev 3 by network update method)

gr3ddy, 2023-11-18:
> [ARK4_Files.zip](https://github.com/PSP-Archive/ARK-4/files/13400985/ARK4_Files.zip)
> 
> Try using both of these.

Yes, it worked👍

closed 2023-11-19 by krazynez