PSP-Archive/ARK-4 #284: firmware update without battery

a0a0max, 2023-12-03:
Original PSP batteries are not working anymore. A lot of users (search for psp update no battery on internet) want update the firmware to 6.60/6.61 avoiding to buy a battery also because they accept the risk to update without battery.

The only working way to update with no battery is Hellcat's Recovery flasher but the maximum allowed Fw Is 6.20. Plugins like FX-Powerspoof are not working for this purpose.

Can you do something?

Thank you all for what you are doing.

closed 2023-12-03 by a0a0max

mtojek, 2024-05-02:
@krazynez What about the case where the battery connector is ripped off and traces are damaged? Technically, the device is not bricked but can't be upgraded/downgraded without the BaryonSweeper. It would be a nice addition to ARK to juggle firmware updates like Chronoswitch does without the battery check.

krazynez, 2024-05-02:
> @krazynez What about the case where the battery connector is ripped off and traces are damaged? Technically, the device is not bricked but can't be upgraded/downgraded without the BaryonSweeper. It [...]

ChronoSwitch does not remove battery checking either. Also you need 6.6X before installing ARK anyways.

mtojek, 2024-05-02:
> Also you need 6.6X before installing ARK anyways

I have PSP Street, 6.60 LME with Infinity.

> ChronoSwitch does not remove battery checking either

Right, that would be a nice patch/feature to disable it. Anyway, I hoped to see an easy method to escape from Infinity in my condition. It looks like the only method is to go with BaryonSweeper?

krazynez, 2024-05-02:
> I have PSP Street, 6.60 LME with Infinity.
>
>
> Right, that would be a nice patch/feature to disable it. Anyway, I hoped to see an easy method to escape from Infinity in my condition. It looks like the [...]

Correct.