PSP-Archive/ARK-4 #285: Psp ISO titles incompatible

Slicer32, 2023-12-15:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.66 Rev 7

### Homebrew/GAME (if applicable)

ULUS-10528 (Gladiator Begins) and homebrew: Nazi Zombies portable github

### Plugins ( if applicable )

Noumd plugins, categories lite plugin

### Steps to reproduce

1.Launch ark 4 launcher
2. Select and start "Gladiator begins" /Nazi zombies portable
3. Psp shutsdown.

Confirmed that both games are working fine on cfw Minimum Edition 2.3 (v6.60). Both being my favourite. 

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

On

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Alway

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2023-12-15:
Do they only crash with the custom launcher? Do they work fine with the XMB? In which format is the Gladiator Begins ISO? As for Nazi Zombies Portable, try setting "Force Extra RAM" to "Homebrew" mode.

krazynez, 2023-12-15:
Can you tell me what you have `categories lite plugin` set to as well please?

Slicer32, 2023-12-16:
The Gladiator Begins both crash in xmb and launcher. All my other 10 games worked flawlessly. I have resolved the Nazi zombies portable thanks to your suggestion.  As for the plugin, i have set using vsh only( folders named for PSX, ISO, Homebrew)

JoseAaronLopezGarcia, 2023-12-16:
> The Gladiator Begins both crash in xmb and launcher. All my other 10 games worked flawlessly. I have tested Nazi Zombies using "homebrew" mode for extra RAM still caused my psp to shutdown. As for the plugin, i have set using vsh only( folders named for PSX, ISO, Homebrew)

Can you tell me where you downloaded Nazi Zombies Portable from?
As for the game, in which format is it? CSO, ISO or PBP?

Slicer32, 2023-12-16:
Gladiator begins is in ISO format, . I downloaded Nazi Zombies portable from github, the latest build by NZP team.

JoseAaronLopezGarcia, 2023-12-16:
There are two versions, the 32MB and the 64MB versions. Which one did you use?

Slicer32, 2023-12-16:
64mb

JoseAaronLopezGarcia, 2023-12-16:
This is strange, Gladiator Begins works fine for me, and so does NZP even with the "Force Extra Memory" option disabled.

Perhaps the files are corrupted? It happens a lot to me, if the USB cable isn't very tight, or maybe I forget to safely remove the device before unplugging, I get games that look ok but crash when launching it.

Slicer32, 2023-12-16:
Both files are good since they play fine using cfw Minimum Edition 2.3. not corrupted.

JoseAaronLopezGarcia, 2023-12-16:
What plugins do you have installed?

krazynez, 2023-12-16:
Yeah try starting both and hold start will booting the games and see if they load. If so a plugin is causing issues.

Slicer32, 2023-12-16:
I have the noumd plugin and categories lite v1.6. Just disabled them and able to run Gladiator Begins by holding the start button. Tried launching the game without holding the start button it boots the psp back to the launcher.

JoseAaronLopezGarcia, 2023-12-16:
> I have the noumd plugin and categories lite v1.6. Just disabled them and able to run Gladiator Begins by holding the start button. Tried launching the game without holding the start button it boots the psp back to the launcher.

The noumd plugin is buggy, use the built-in "Disable UMD Drive" option that ARK already has.

Slicer32, 2023-12-17:
Ok, I did try disabling noumd plugin, now only my categories lite v1.6 is enabled. Its finicky by holding start to launch a game for it to work though. As for nazi zombies homebrew i can't get it running regardless of the settings I tried. For the timebeing I'm just using cfw Minimum Edition 2.3 for stability and compatibility purposes. Btw i did disable the UMD on Ark settings. Thanks guys for the suggestions.

krazynez, 2023-12-17:
Can you upload your PLUGINS.TXT 

located: ms0:/PSP/SAVEDATA/ARK-01234/PLUGINS.TXT

Slicer32, 2023-12-17:
ms0:/seplugins/categorieslite/category_lite.prx 1

btw i did not fully installed ark 4(not yet flash cipl and ark 4 full installation) since I'm still new to it and testing.

krazynez, 2023-12-17:
> ms0:/seplugins/categorieslite/category_lite.prx 1
> 
> btw i did not fully installed ark 4(not yet flash cipl and ark 4 full installation) since I'm still new to it and testing.

That is not how our plugins work

Slicer32, 2023-12-17:
exactly, i don't know how to show it here, the "on" part, so far that's the only enabled plugin i have

krazynez, 2023-12-17:
> exactly, i don't know how to show it here, the "on" part, so far that's the only enabled plugin i have. Sometimes when i exit the launcher it goes to vanilla mode without any options to access the launcher and vsh menu(select button). As for the gladiator game it shows "game could not be started (80010016) when i launched it in xmb vanilla mode

rigght but your syntax for your plugin is how PRO & ME use it not ARK-4 it should never be working that way with ARK-4 running.

Sounds like you might be rebooting instead of exiting. So with ARK-4 Live its only temp running so if you reboot/reset the PSP ARK-4 is not loaded anymore. cIPL would fix that however.

Slicer32, 2023-12-17:
yes i"m temp running since i haven't run cipl and permanent install Ark 4. Would permanent install via Cipl be reversible though(uninstall) in case I still run with incompatible issues. I might give it ago for full installation.

JoseAaronLopezGarcia, 2023-12-17:
> Ok, I did try disabling noumd plugin, now only my categories lite v1.6 is enabled. Its finicky by holding start to launch a game for it to work though. As for nazi zombies homebrew i can't get it [...]

You don't have to hold start all the time, holding start disables all plugins, so that means there was a plugin that was causing it to crash, which I suspect it was the NoUMD plugin as that one is known to cause issues, try launching it now without the start button.

krazynez, 2023-12-17:
> yes i"m temp running since i haven't run cipl and permanent install Ark 4. Would permanent install via Cipl be reversible though(uninstall) in case I still run with incompatible issues. I might give it ago for full installation.

Yeah cipl can be uninstalled just reinstalling OFW again. But I would first get your issues resolved first before doing cIPL stuff. Both Acid and I can't reproduce this issue.

krazynez, 2023-12-18:
I also have now verified that both Gladiaor Begins and NZP both working in Custom Launcher with your settings listed above. They only thing I do not have installed is Game Category Lite. Have you tried to run them again after noumd plugins being removed that @JoseAaronLopezGarcia stated above?

Slicer32, 2023-12-20:
Just reformatted my psp, planning to reinstall ark 4 again to check whether those issues are still present or otherwise, should i download the latest "pre-release" version or the release version on October 24th?

Slicer32, 2023-12-20:
> I also have now verified that both Gladiaor Begins and NZP both working in Custom Launcher with your settings listed above. They only thing I do not have installed is Game Category Lite. Have you tried to run them again after noumd plugins being removed that @JoseAaronLopezGarcia stated above?

PS: I did run it with all plugins disabled in both vanilla xmb and the launcher itself, resulting the psp to shutdown and reboot back to ME 2.3. I guess I can't run both temporary cfws.

krazynez, 2023-12-20:
> Just reformatted my psp, planning to reinstall ark 4 again to check whether those issues are still present or otherwise, should i download the latest "pre-release" version or the release version on October 24th "Version 4.20.66 Rev 7"?

Stable version is fine.

Slicer32, 2023-12-28:
Alright guys, after 2 weeks of review, reformatting, and only using ark 4( no ME 2.3 or pro C fix 3) i can definitely point out the issues of compatibility that I previously faced are no more. Thanks for the feedback, suggestions and constant guidance. May this project becomes the best cfw the psp ever has.

krazynez, 2023-12-28:
> Alright guys, after 2 weeks of review, reformatting, and only using ark 4( no ME 2.3 or pro C fix 3) i can definitely point out the issues of compatibility that I previously faced are no more. [...]

Awesome to hear. I will close the issue then. Thanks for the update.

closed 2023-12-28 by krazynez