PSP-Archive/ARK-4 #286: Time played counter.

agentbarti, 2023-12-18:
### New Feature

Hello, I've just thought of a cool feature. That will count time spent in certain games. If it's doable it will display in PSP'S xmb.

krazynez, 2023-12-18:
XMB probably not. Custom Launcher probably would be more likley.

krazynez, 2023-12-18:
Okay so the main dev and I spoke about this and we both agree that if this was more of higher user request we could, but this would increase the ram for all users which we are somewhat already limited it. We do appreciate the request. Possibly someone can make a plugin to do that (that way you could also have it on XMB as well more easily). Thank you  ;-)

closed 2023-12-18 by krazynez

JoseAaronLopezGarcia, 2023-12-18:
Yeah this should be done as a plugin, not a built-in CFW feature.