PSP-Archive/ARK-4 #288: plugin XMB

ZweiBR, 2024-01-01:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

Version 4.20.66 Rev 7

### Steps to reproduce

https://github.com/PSP-Archive/ARK-4/raw/main/.github/screenshots/xmb(2).png

I would like to know where this option is in the configuration to turn plugins on and off?

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

ZweiBR, 2024-01-01:
sorry I found the option
located in the EXTRAS tab

krazynez, 2024-01-02:
Ahh need to update screenshots for it sorry about that.

closed 2024-01-02 by krazynez

ZweiBR, 2024-01-02:
> Ahh need to update screenshots for it sorry about that.

Thank you so much for your wonderful work, congratulations.

ZweiBR, 2024-01-02:
I wanted to know if ARK-4 supports the cxmb plugin? I want to test themes in CTF format.

krazynez, 2024-01-02:
> I wanted to know if ARK-4 supports the cxmb plugin? I want to test themes in CTF format.

yes it does.

ZweiBR, 2024-01-02:
Version 4.20.66 Rev 7
I have a suggestion for numbering your CFW.
It would be possible to change to 6.61-0 and so on.
6.61-1, 6.61-2, 6.61-3 etc.

JoseAaronLopezGarcia, 2024-01-02:
> Version 4.20.66 Rev 7 I have a suggestion for numbering your CFW. It would be possible to change to 6.61-0 and so on. 6.61-1, 6.61-2, 6.61-3 etc.

6.61 isn't the only supported firmware though, 6.60 is also supported and all of Vita's ePSP firmwares that are based on 6.60 but are their own thing.