PSP-Archive/ARK-4 #290: Random Theme on startup for XMB

DeadLootJ, 2024-01-11:
### New Feature

The idea is to have a theme (.ptf) randomly selected on startup/ VSH reset. This can be done on CXMB with random.ctf. I was curious if something similar could be done with the standard XMB themes. Maybe as part of the Custom Firmware Settings or the Advanced VSH Menu?

JoseAaronLopezGarcia, 2024-01-11:
Wouldn't that be VERY slow? Since it'd have to apply a new theme on every boot and every time you exit a game.

DeadLootJ, 2024-01-11:
On CXMB, with random.ctf, the startup/reset times are about the same as standard XMB startup/reset (from my perception). On the standard XMB, exiting from a game appears to not trigger a VSH reset, so I would think there would be no impact there.

DeadLootJ, 2024-01-11:
Ok, I can see how this could potentially become a big problem. An alternative, Advanced VSH Menu has an option to Boot Random ISO. Could there be an option to apply a random theme here? Or would this lead back to flash storage breaking as well?

krazynez, 2024-01-11:
> Ok, I can see how this could potentially become a big problem. An alternative, Advanced VSH Menu has an option to Boot Random ISO. Could there be an option to apply a random theme here? Or would this lead back to flash storage breaking as well?

That would still lead the same path (corruption of flash) possibly.

DeadLootJ, 2024-01-11:
I suppose that could lead to that with overuse. On a related note, I should probably calm down on my theme swapping. I change my theme at least once a day, to every few days. Dependant upon how often I get on the PSP.

JoseAaronLopezGarcia, 2024-01-11:
> Ok, I can see how this could potentially become a big problem. An alternative, Advanced VSH Menu has an option to Boot Random ISO. Could there be an option to apply a random theme here? Or would this lead back to flash storage breaking as well?

well if it's only that happens when you press the "Random Theme" option instead of every time you boot into VSH then I see no problem.

krazynez, 2024-01-11:
> well if it's only that happens when you press the "Random Theme" option instead of every time you boot into VSH then I see no problem.

Ahhh I was thinking it was like the colors, yeah that could work also then.

wad11656, 2025-02-04:
Well there's Wallpaper Changer, which _at least_ allows you to change the wallpaper each time you enter the XMB, but I believe you can also set a timer to change the wallpaper (though I haven't gotten this feature to work).

https://www.gamebrew.org/wiki/WallpaperChanger_Mod_PSP

JoseAaronLopezGarcia, 2025-03-16:
We will not be doing this so I'm closing it.