PSP-Archive/ARK-4 #291: Beowulf - The Game not working black screen after psp logo

pitrako, 2024-01-13:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

ARK 4.20.66 r7

### Homebrew/GAME (if applicable)

Beowulf - The Game ULES-00992

### Plugins ( if applicable )

No plugins

### Steps to reproduce

1- Start game.
2- Wait after PSP logo.
3- Black screen.
4- Error C1-2858-3

### USB Charge

Disabled

### Overclock

Game

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Homebrew

### Memory Stick Speedup

Alway

### Inferno Cache

RR

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-01-13:
Is this a standalone installation or an Adrenaline-ARK installation?

pitrako, 2024-01-13:
I tried holding select and manually disabling inferno cache on custom firmware settings section and nothing changed.

JoseAaronLopezGarcia, 2024-01-13:
What's the format? ISO, CSO or PBP?

pitrako, 2024-01-13:
It's in ISO format.

krazynez, 2024-01-13:
[image]
US version same settings. I will also test EU version.

JoseAaronLopezGarcia, 2024-01-13:
If it doesn't work on regular Adrenaline either then that means only one thing, the ISO is corrupted. Try copying it again.

krazynez, 2024-01-13:
Actually can replicated it for EU version. Testing a lot of settings to see if there is something that will allow it to boot.

pitrako, 2024-01-13:
USA works:
[image]

EUR not working (sorry upside down, my phone acting weird, It says an error ocurred c1-2858-3.)
[image]

JoseAaronLopezGarcia, 2024-01-13:
Does it work with a Standalone installation?

pitrako, 2024-01-13:
Nope, not working in Standalone.

krazynez, 2024-01-13:
tested as an EBOOT with same results on Standalone as well (not working)

pitrako, 2024-06-24:
Any news on this?