PSP-Archive/ARK-4 #292: Eyepet camera (back and front) not working, empty black photo, laggy intro cutscene, audio not synced

pitrako, 2024-01-14:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

ARK 4.20.66 r8 Standalone

### Homebrew/GAME (if applicable)

Eyepet UCES-01462

### Plugins ( if applicable )

No plugins

### Steps to reproduce

1-Start game, first cutscene laggy and audio not synced.
2-It asks to take a photo after cinematic.
3-Photos are empty with black, the camera is not covered:
EUR
[image]
USA
[image]
4-If I change camera to front or back It crashes with error:
[image]
5-On a PSP1000 It asks to connect a camera (I don't have one), on PSVITA not even asking because camera already present but displaying nothing:
PSP1000
[image]

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-01-16:
This is an excellent bug report. I wonder if the camera is somehow locked away if the "game" (in this case, ARK's loader) doesn't match some known ID in some database.
I know this is done for USB and PocketStation support, but no idea about camera.

j0s3ph19, 2025-03-09:
i had the same issue too, but in the XMB camera works...

JoseAaronLopezGarcia, 2025-03-16:
Seems to be a limitation of the vita when it emulates camera. We're not too sure if it's fixable.
More information here: https://github.com/PSP-Archive/ARK-4/issues/502

closed 2025-03-16 by JoseAaronLopezGarcia

ItsBerna, 2025-07-23:
So there's no hope of this ever working?😫

JoseAaronLopezGarcia, 2025-07-23:
> So there's no hope of this ever working?😫

The thing is that this isn't an ARK issue, it's just how pspemu works. Sony did the bare minimal for cam and mic.
There has been attempts at fixing this but it is not simple.