PSP-Archive/ARK-4 #296: God Eater 2 (English Patched) "PlayStation Network" message on screen

Mecario, 2024-01-20:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.67 r2

### Homebrew/GAME (if applicable)

NPJH-50832

### Steps to reproduce

Simply starting the game as usual 
Additional note: issue not present on PRO-C2

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-01-24:
[image]

Working fine on my 3K.

JoseAaronLopezGarcia, 2024-01-24:
Perhaps it needs extra RAM enabled? Or some other setting.

krazynez, 2024-01-24:
> Perhaps it needs extra RAM enabled? Or some other setting.

I have nothing enabled on my 3K just brought it back to 6.61 to test so fresh stable release of ARK is installed.

JoseAaronLopezGarcia, 2024-01-24:
Could it have anything to do with the console's region?

Mecario, 2024-01-24:
I have an european console in case that helps

JoseAaronLopezGarcia, 2024-02-03:
What about the firmware version?

krazynez, 2024-02-03:
[image]
What it translates to, did you keep your update file for the game that are required?

Mecario, 2024-02-03:
I went back to the official european 6.61 using chronoswitch, then ran ark4 and made it permanent with infinity, so it reads as follows: 6.61 ARK-4 Infinity∞

JoseAaronLopezGarcia, 2024-02-03:
Perhaps he needs nploader to load the update?

Mecario, 2024-02-03:
> ![image](https://private-user-images.githubusercontent.com/39999447/302044272-b04ea850-92b6-4799-8ab3-83ed67b1398f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I'm pretty sure I did and it works perfectly on another psp3000 I have with proc2 (same card, so I thought it had to be on ark4's side). But I'll try to set it up again just in case.

closed 2024-02-07 by Mecario