PSP-Archive/ARK-4 #297: [PSP go] Games loaded from internal storage in the custom launcher will load savedata from memory stick instead of internal storage

koktagon, 2024-01-24:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.67 r3

### Homebrew/GAME (if applicable)

ULUS-10160

### Steps to reproduce

1. Go into the custom launcher
2. Start an ISO game from internal storage (tested with GTA Vice City Stories)
3. Load savedata

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-01-25:
I think you have to enable "Redirect ms0 to ef0" in the launcher's settings.

JoseAaronLopezGarcia, 2024-02-03:
Closing as this is not really a bug, but an optional feature, Custom Launcher lets you choose the main device for savedata when launching ISO or homebrew from ef0 (does NOT apply to PSN and PS1 games).

closed 2024-02-03 by JoseAaronLopezGarcia