PSP-Archive/ARK-4 #298: Simulation of ps1 game crashes

kingboxnt, 2024-01-31:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

6.60-4.20.67 Rev 5

### Homebrew/GAME (if applicable)

PS1: SLPS03177

### Plugins ( if applicable )

CheatMaster_R21d_CN
[CMF_R21d_CN.zip](https://github.com/PSP-Archive/ARK-4/files/14106888/CMF_R21d_CN.zip)

### Steps to reproduce

Random crash when opening CMF_R21d_CN game.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-01-31:
How is the plugin installed? Instructions online mention enabling it for game, pops and vsh (so always), but a cheat plugin on vsh doesn't make much sense to me, but who knows.
I'll give it a try and let you know.

krazynez, 2024-01-31:
[image]
Works fine on 1K, just they're no cheats in the db for that title. Even checked the massive database for the gameid with no results. If you can verify which db you are using that has that gameid I can look further into it.

kingboxnt, 2024-01-31:
> ![image](https://private-user-images.githubusercontent.com/39999447/301175456-c07065cc-f4bd-421f-bf75-812ecef77764.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I'm not using any cheat codes, just the resident plugin, and it doesn't cause a problem at the beginning of the credits, but when I enter the first level of the game, and try to play it crashes randomly, nowhere in particular, same with SLPS03050.

me and pro-c are fine.

krazynez, 2024-01-31:
[image]
Still working, though initially I had to run it first disabling the plugin (holding Start when launching) or it would kick me back to CL (Custom Launcher) but afterwards its working still.

JoseAaronLopezGarcia, 2024-02-17:
Any idea?

kingboxnt, 2024-02-18:
> Any idea if this is fixed?

I replaced the ips screen, psp is temporarily paralyzed, can't try until the new screen arrives in a few days.

kingboxnt, 2024-02-22:
It's resting in peace. I give up.

closed 2024-02-22 by kingboxnt

JoseAaronLopezGarcia, 2024-02-22:
> It's resting in peace. I give up.

Why, is the console broken?

kingboxnt, 2024-02-23:
> > It's resting in peace. I give up.
> 
> Why, is the console broken?

Replacement of ips screen need to solder, actually replace the first piece of ips screen successfully, but in the wrap insulation tape wrapped the wrong conductive tape, resulting in a short, and now the second piece of ips screen connected to the blinking green light, the motherboard should be bad, laugh!