PSP-Archive/ARK-4 #301: Standalone version 4.20.67 Rev 6 crash when HUD enabled and on exit game hotkey

pmsobrado, 2024-02-04:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.67 Rev 6

### Homebrew/GAME (if applicable)

Any

### Plugins ( if applicable )

Tried with

"game, ms0:/seplugins/hud.prx, on"

and

"game, ms0:/seplugins/hud.prx, on"

### Steps to reproduce

Enable HUD in plugins.txt. Open standalone ARK, it should crash.

Disable HUD in plugins.txt. Open standalone ARK, now it does not crash, launch any game, press hotkeys to exit and back to launched, it should crash.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-02-04:
[image]

Latest version of ARK-4 running via Standalone

pmsobrado, 2024-02-04:
Damn, something's wrong with mine then.

@krazynez when in-game, with HUD plugin enabled (not necessarily visible) can you go back to launcher with the exit hotkeys?

krazynez, 2024-02-04:
> Damn, something's wrong with mine then.
> 
> @krazynez when in-game, with HUD plugin enabled (not necessarily visible) can you go back to launcher with the exit hotkeys?

That is indeed not working, guessing HUD is looking at all button presses overriding ARK-4 Exit Mask.

JoseAaronLopezGarcia, 2024-02-04:
Right, I checked last time and found that some plugins like this one will override the controller patch that ARK applies.
I have one possible solution for this.

pmsobrado, 2024-02-04:
@krazynez OK, HUD plugin now works for me, it must be another plugin incompatibility, never mind about that.

@JoseAaronLopezGarcia As for the hotkey to go back to the launcher, without HUD plugin enabled, it is working fine. With HUD plugin, it does not, so I guess we can close this issue and keep the conversation at https://github.com/PSP-Archive/ARK-4/issues/268

As the title says, very minor issue, so don't worry to much :)

closed 2024-02-04 by JoseAaronLopezGarcia