PSP-Archive/ARK-4 #302: TwinBee Portable doesn't work unless system is set to english language.

Twinbeeman, 2024-02-05:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

Version 4.20.67 Rev 5

### Homebrew/GAME (if applicable)

ULJM-05221

### Steps to reproduce

Change the system language to anything that is not English or Japanese and the game crashes right as you launch it. 
PS. This is a known issue within the game itself so I didn't really know if it should be labeled as a bug or a feature request, tbh.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-02-05:
It should be possible to patch the game so it falls back to english language when the system is set to anything other than english or japanase. Would at least prevent the crash.

JoseAaronLopezGarcia, 2024-02-07:
Fixed added to latest release.

closed 2024-02-07 by JoseAaronLopezGarcia