PSP-Archive/ARK-4 #303: No games found in CLauncher

Jambul46, 2024-02-06:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.68

### Plugins ( if applicable )

always, ms0:/SEPLUGINS/battWarn/BATTWARN.PRX, on

xmb, ms0:/SEPLUGINS/categorieslite/category_lite.prx, on

xmb, ms0:/SEPLUGINS/CFE/cfe_loader.prx, on

game, ms0:/SEPLUGINS/CFE/cfe_loader.prx, on

game, ms0:/SEPLUGINS/CWCHEAT/CWCHEAT.PRX, on

ps1, ms0:/SEPLUGINS/CWCHEAT/cwcheatpops.prx, on

game, ms0:/SEPLUGINS/MP3PLAY/MP3PLAY.PRX, on

xmb, ms0:/SEPLUGINS/PPS/PPS.PRX, on

always, ms0:/SEPLUGINS/PPS/PPS.PRX, on

xmb, ms0:/SEPLUGINS/WP/WP.PRX, on

xmb, ms0:/SEPLUGINS/XMBIH/XMBIH.PRX, off

homebrew, ms0:/SEPLUGINS/LEDA/leda.prx, on

### Steps to reproduce

Start CLauncher with Category Lite folder system (CAT_XXxxxxx)

### USB Charge

Always

### Overclock

Homebrew

### Power Save

Disabled

### Balanced Energy Mode

Game

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-02-06:
Did you enable the setting in in CL settings to scan caetegories ?

Jambul46, 2024-02-06:
Also yeah thanks i forgot i accidentaly reset my settings.

closed 2024-02-06 by Jambul46