PSP-Archive/ARK-4 #305: Psp 256gb+ memory stick support through micro sd adapter

Mecario, 2024-02-07:
### New Feature

I was wondering if it would be possible to achieve this through ark4. Any micro sd card over 128gb won't be detected or formated from a psp on itself, doing this through usb with a 256gb card inside will result on it being limited to 104gb (instead of the 116~gb usable space you would usually get from a 128gb card). I've tried this with multiple micro sds, adapters and a couple psp 3k and always obtained the same results. At one point I even managed to get the psp to detect the full size, but this would still be limited to files up to 104gb nonetheless, anything over it could simply not be added through usb (connecting the micro sd directly  to a computer would allow to transfer more games but these would then appear as corrupted on the psp side only). As far as I know this issue doesn't affect the ps vita.

krazynez, 2024-02-07:
This is not and ARK-4 or any CFW issue its the memory stick adapters limitation.

Mecario, 2024-02-07:
Does that mean it would be possible in theory if an adapter without that limitation was made?

krazynez, 2024-02-07:
Yes but your never going to get full disk space there are a lot of things that go into account with that.

Mecario, 2024-02-07:
I see, thank you for the information

wad11656, 2024-09-01:
> This is not and ARK-4 or any CFW issue its the memory stick adapters limitation.

I have a MicroSD slot directly soldered onto my PSP GO's motherboard, and the same limitations of only being able to format cards that are <=128GB occurs. What are the chances that every single microSD adapter, including a soldered-on MicroSD slot, all have this limitation hardcoded into them? I strongly doubt the metal piece that was soldered onto my motherboard (which wasn't even manufactured for the PSP in particular) has a microcontroller inside it that dictates that it only supports 128gb or less. It really seems like a software issue to me. I think this issue warrants more thorough testing and investigation. (Not that I personally care too much, because without using the categories lite plugin, more than 128gb is very sluggish)

In the video below, he uses a 1TB MicroSD card plugged into one of these MicroSD-MemoryStick Duo adapters on both his PSP and his PC. On the PSP, it shows a software-dictated size limit. On the PC, it does not: 

https://youtu.be/k76EmFxnYWo?si=9YZXNvi3bfvZUbMq&t=74

After cracking some of these adapters open, it looks like the adapters are simply rerouting pins--there's no microcontrollers inside them that are limiting the amount of storage they support.

hobb7t, 2026-01-09:
> > This is not and ARK-4 or any CFW issue its the memory stick adapters limitation.
> 
> I have a MicroSD slot directly soldered onto my PSP GO's motherboard, and the same limitations of only being able to format cards that are <=128GB occurs. What are the chances that every single microSD adapter, including a soldered-on MicroSD slot, all have this limitation hardcoded into them? I strongly doubt the metal piece that was soldered onto my motherboard (which wasn't even manufactured for the PSP in particular) has a microcontroller inside it that dictates that it only supports 128gb or less. It really seems like a software issue to me. **I think this issue warrants more thorough testing and investigation.** I personally don't know if I would utilize a workaround to use cards beyond 128gb, but I know it's a fairly frequently-asked question, at least on Reddit.
> 
> In the video below, he uses a 1TB MicroSD card plugged into one of these MicroSD-MemoryStick Duo adapters on both his PSP and his PC. On the PSP, it shows a software-dictated size limit. On the PC, it does not: 
> 
> https://youtu.be/k76EmFxnYWo?si=9YZXNvi3bfvZUbMq&t=74
> 
> After cracking some of these adapters open, it looks like the adapters are simply rerouting pins--there's no microcontrollers inside them that are limiting the amount of storage they support.

This seems to be the truth, especially when everybody just drop in some sort of answer like "this is an adapter's limitation" with no facts or any legitimate source that proves such a limitation. The adapter does nothing, except doing its job, or being garbage, but psp can not work with larger SDs, you can even get a timeout error trying to format such a card while plugged. Probably due to limited resources on PSP side