PSP-Archive/ARK-4 #306: ARK-4 forces me to format the PS Vita

Databoi2023, 2024-02-08:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.68 Rev 3

### Steps to reproduce

So, after making a clean ARK-4 install, and running VitaShell (installed by VitaDeploy's App downloader), I have the C2-12828-1 error. After clicking "OK", the PS Vita freezes with a black screen. If I access Safe Mode to restart it or rebuild its database, the PS Vita has spinning white dots until they freeze. I have no other choice but to format it using the Safe Menu and to reinstall all my games and homebrew. I also have this issue on the PS TV.

I just jailbroke the PS Vita with the HENlo exploit, downgraded to 3.60, chose the "HEN only" taiHEN configuration and checked the "Install enso" setting in VitaDeploy, installed VitaShell using the latter, installed AutoPlugin II and its NoPspEmuDrm plugin before installing FasterARK and ARK-4 and running VitaShell. I've done multiple PS Vita restores using the Safe Mode and I always have this problem after following these steps.

Here's my PS Vita's plugin configuration before installing FasterARK:

# This file is used as an alternative if ux0:tai/config.txt is not found.
# For users plugins, you must refresh taiHEN from HENkaku Settings for
# changes to take place.
# For kernel plugins, you must reboot for changes to take place.
*KERNEL
ur0:tai/NoPspEmuDrm_kern.skprx
# henkaku.skprx is hard-coded to load and is not listed here
*main
# main is a special titleid for SceShell
ur0:tai/henkaku.suprx
*NPXS10015
# this is for modifying the version string
ur0:tai/henkaku.suprx
*NPXS10016
# this is for modifying the version string in settings widget
ur0:tai/henkaku.suprx
*ALL
ur0:tai/NoPspEmuDrm_user.suprx

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-02-08:
There's something strange about your plugins config and your bug report.

You mention using 4.20.68 Rev 3, by this point the FasterARK version you're using should be installing both ARK and ARK-X, so you should be seeing two ARK bubbles (one for PSP, one for PS1). Do you see both bubbles or only one?

The problem I see is that the `ps1cfw_enabler.suprx` plugin isn't configured in your `config.txt`, which should have been done by FasterARK. I'm not sure if it's a bug, or perhaps you're using a slightly older revision instead of Rev 3.

Databoi2023, 2024-02-08:
> There's something strange about your plugins config and your bug report.
>
> You mention using 4.20.68 Rev 3, by this point the FasterARK version you're using should be installing both ARK and ARK-X, [...]

Only once, since I removed ARK-X because I don't want it. Will there be an option to not install ARK-X?

No, I tried minutes ago. Also, I have this problem even if not removing ARK-X.

JoseAaronLopezGarcia, 2024-02-08:
Yeah you can get rid of ARK-X, it won't be an issue. I find it strange that ARK causes native vita homebrew to crash considering that ARK is a PSP homebrew.

Databoi2023, 2024-02-08:
> Yeah you can get rid of ARK-X, it won't be an issue. I find it strange that ARK causes native vita homebrew to crash considering that ARK is a PSP homebrew.

Me too. I've never had an issue as serious as this one with an app. Could you follow my steps from start to finish?

JoseAaronLopezGarcia, 2024-02-08:
I just recently used FasterARK from scratch without any issues, and I also have a lot more plugins installed than you.

Databoi2023, 2024-02-08:
> I just recently used FasterARK from scratch without any issues, and I also have a lot more plugins installed than you.

Then I don't know. I also have this issue on the PS TV. Did you jailbroke the PS Vita with the HENIo exploit, downgraded to 3.60, chose the "HEN only" taiHEN configuration and checked the "Install enso" setting in VitaDeploy, installed VitaShell using the latter's App downloader, installed AutoPlugin II and its NoPspEmuDrm plugin, installed FasterARK and ARK-4 (not opening ARK-4 just yet) and ran VitaShell?

JoseAaronLopezGarcia, 2024-02-08:
I installed henkaku via the henkaky.xyz page (web browser exploit). I don't remember choosing any "HEN only" option. I didn't use VitaDeploy, I used the original installers from the original website.

In any case, this doesn't seem to be an ARK issue, since ARK doesn't really have any Vita-side code AT ALL, it can't possibly be the cause of your issues.

I would advise to just reinstall Henkaku from scratch using the official website and tools.

Databoi2023, 2024-02-08:
> I installed henkaku via the henkaky.xyz page (web browser exploit). I don't remember choosing any "HEN only" option. I didn't use VitaDeploy, I used the original installers from the original [...]

I thought jailbreak.psp2.dev was the official HENIo exploit site and I only have this issue after installing ARK-4. 

What are the official website (I guess henkaky.xyz that you mentioned) and tools, then? Did you mean "henkaki.xyz" or "henkaku.xyz"? The "HEN only" option doesn't appear in the site I mentioned but in VitaDeploy. I followed [this](https://m.youtube.com/watch?v=uWXJgX5rgTw) video but chose another taiHEN configuration (the "HEN only").

Does the site you mentioned (or henkaku.xyz) install some apps along HENkaku, and if so, which ones?

JoseAaronLopezGarcia, 2024-02-08:
I did not use henlo cause all my vita's were already in 3.60 where you can use the original henkaku browser exploit.

Databoi2023, 2024-02-08:
> I did not use henlo cause all my vita's were already in 3.60 where you can use the original henkaku browser exploit.

In that case, how can 3.74 OFW be downgraded?

JoseAaronLopezGarcia, 2024-02-08:
You can use Modoru (I think that's the name) to downgrade to 3.60 or 3.65, which are by far a lot more stable and compatible.

krazynez, 2024-02-08:
I used vitawiki.xyz/henlo/ on 3.65 to install Henkaku works fine

Pretty sure henkaku.xyz only supports 3.60

Databoi2023, 2024-02-08:
> You can use Modoru (I think that's the name) to downgrade to 3.60 or 3.65, which are by far a lot more stable and compatible.

It seems the exploit you mentioned doesn't install VitaDeploy which is very useful to install apps and installs an app called molecular shell. Is it related to VitaShell?

Also, if following your web exploit, how can enso be installed to make HENkaku permanent?

And what if following the [previous web exploit I tried](https://m.youtube.com/watch?v=KUMQFEiuDJ4), the one which installs an app called h-encore² and VitaDeploy? Do you think it would have this problem as well because I'd follow the same VitaDeploy steps?

Anyway, in that case, could you alert users to follow the exploit you mentioned and not mine since it apparently causes this issue or follow my steps so that you may know if this also happens to you?

Databoi2023, 2024-02-08:
> I used vitawiki.xyz/henlo/ on 3.65 to install Henkaku works fine
> 
> Pretty sure henkaku.xyz only supports 3.60

Can any of you try jailbreak.psp2.dev and follow my remaining steps to be sure I'm not the only having this problem, please?

JoseAaronLopezGarcia, 2024-02-08:
Sorry but none of my PS Vita's are on 3.74, they are all on 3.60, and most homebrew support is on 3.60 and 3.65 anyways so there's not much of a reason for us to go to later firmwares.

Databoi2023, 2024-02-08:
> Sorry but none of my PS Vita's are on 3.74, they are all on 3.60, and most homebrew support is on 3.60 and 3.65 anyways so there's not much of a reason for us to go to later firmwares.

Then why not upgrade do 3.74, downgrade to 3.60 using the tool you mentioned and follow my steps?

JoseAaronLopezGarcia, 2024-02-08:
Why would I wanna do that? What would I win? I already told you that most vita homebrew support `3.60` and `3.65`, not 3.74, I don't need to update to know that 3.74 has limited compatibility, it's a well known fact.
Why you keep insisting on 3.74 at this point is beyond me, everyone simply downgrades to either `3.60` or `3.65`.

Databoi2023, 2024-02-08:
> Why would I wanna do that? What would I win? I already told you that most vita homebrew support `3.60` and `3.65`, not 3.74, I don't need to update to know that 3.74 has limited compatibility, it's [...]

Because that way you'd know if ARK-4 causes this issue since the question isn't 3.74 but using the HENlo exploit and to reproduce my steps you'd have to upgrade to 3.74. Besides, how could I use that downgrade tool you mentioned in OFW 3.74 since my system wouldn't even be hacked in the first place?

JoseAaronLopezGarcia, 2024-02-08:
Like I said, ARK-4 itself doesn't touch any native vita side, it's purely a PSP software. This is probably due to a plugin misbehaving. I'm not sure but I believe NoPspEmuDrm doesn't support 3.74, or maybe some other app or plugin.

You could try to use ChovySign to create your own ARK-4 bubble, which would work even on Official Firmware without henlo or henkaku.

Databoi2023, 2024-02-08:
> Like I said, ARK-4 itself doesn't touch any native vita side, it's purely a PSP software. This is probably due to a plugin misbehaving. I'm not sure but I believe NoPspEmuDrm doesn't support 3.74, [...]

Like I said, I downgraded to 3.60 so I'm not in 3.74. Also, like I should you, I barely have any plugins so how can they cause this issue?

krazynez, 2024-02-08:
> Because that way you'd know if ARK-4 causes this issue since the question isn't 3.74 but using the HENlo exploit and to reproduce my steps you'd have to upgrade to 3.74. Besides, how could I use [...]

You can downgrade from 3.74 to 3.60 or 3.65 via Modoru.

Databoi2023, 2024-02-08:
> You can downgrade from 3.74 to 3.60 or 3.65 via Modoru.

How? It seems it needs HENkaku but the exploit @JoseAaronLopezGarcia mentioned doesn't work in 3.74. Since he doesn't want to follow my steps, what about you?

krazynez, 2024-02-08:
> How? It seems it needs HENkaku but the exploit @JoseAaronLopezGarcia mentioned doesn't work in 3.74. Since he doesn't want to follow my steps, what about you?

Ahh my mistake was thinking that was a webexploit way here:
use HENkaku on 3.74 once that is running use Modoru to downgrade to 3.60 3.65

Databoi2023, 2024-02-08:
> Ahh my mistake was thinking that was a webexploit way here: use HENkaku on 3.74 once that is running use [Modoru](https://github.com/TheOfficialFloW/modoru) to downgrade to 3.60 3.65

But the question is, how can I install HENkaku without following the HENl exploit?

krazynez, 2024-02-08:
> > > > > > Why would I wanna do that? What would I win? I already told you that most vita homebrew support `3.60` and `3.65`, not 3.74, I don't need to update to know that 3.74 has limited compatibility, it's a well known fact. Why you keep insisting on 3.74 at this point is beyond me, everyone simply downgrades to either `3.60` or `3.65`.
> > > > > 
> > > > > 
> > > > > Because that way you'd know if ARK-4 causes this issue since the question isn't 3.74 but using the HENlo exploit and to reproduce my steps you'd have to upgrade to 3.74. Besides, how could I use that downgrade tool you mentioned in OFW 3.74 since my system wouldn't even be hacked in the first place?
> > > > 
> > > > 
> > > > You can downgrade from 3.74 to 3.60 or 3.65 via Modoru.
> > > 
> > > 
> > > How? It seems it needs HENkaku but the exploit @JoseAaronLopezGarcia mentioned doesn't work in 3.74. Since he doesn't want to follow my steps, what about you?
> > 
> > 
> > Ahh my mistake was thinking that was a webexploit way here: use HENkaku on 3.74 once that is running use [Modoru](https://github.com/TheOfficialFloW/modoru) to downgrade to 3.60 3.65
> 
> But the question is, how can I install HENkaku on 3.74 without following the HENlo exploit?

we didn't say don't do that just don't do that and expect things to work on 3.74. You will still do that to run Modoru with vitashell then run and downgrade.

Databoi2023, 2024-02-08:
> > > > > > > Why would I wanna do that? What would I win? I already told you that most vita homebrew support `3.60` and `3.65`, not 3.74, I don't need to update to know that 3.74 has limited compatibility, it's a well known fact. Why you keep insisting on 3.74 at this point is beyond me, everyone simply downgrades to either `3.60` or `3.65`.
> > > > > > 
> > > > > > 
> > > > > > Because that way you'd know if ARK-4 causes this issue since the question isn't 3.74 but using the HENlo exploit and to reproduce my steps you'd have to upgrade to 3.74. Besides, how could I use that downgrade tool you mentioned in OFW 3.74 since my system wouldn't even be hacked in the first place?
> > > > > 
> > > > > 
> > > > > You can downgrade from 3.74 to 3.60 or 3.65 via Modoru.
> > > > 
> > > > 
> > > > How? It seems it needs HENkaku but the exploit @JoseAaronLopezGarcia mentioned doesn't work in 3.74. Since he doesn't want to follow my steps, what about you?
> > > 
> > > 
> > > Ahh my mistake was thinking that was a webexploit way here: use HENkaku on 3.74 once that is running use [Modoru](https://github.com/TheOfficialFloW/modoru) to downgrade to 3.60 3.65
> > 
> > 
> > But the question is, how can I install HENkaku on 3.74 without following the HENlo exploit?
> 
> we didn't say don't do that just don't do that and expect things to work on 3.74. You will still do that to run Modoru with vitashell then run and downgrade (with following the steps that the link provides how to use Modoru).

But like I said, when I have this problem I'm not in 3.74 but in 3.60. Acoording to the link you showed, to use modoru HENkaku needs to be installed as well as VitaShell. Again, how can I install HENkaku on 3.74 in the first place?

JoseAaronLopezGarcia, 2024-02-08:
So is your vita on 3.74 or 3.60? I think you just used the wrong henkaku version for your firmware. Try to uninstall enso and henkaku and reset your vita.

ARK-4 works on 3.74, it's been tested before. The errors you're getting are not related to ARK but a bad installation of the hacks.

Databoi2023, 2024-02-08:
> So is your vita on 3.74 or 3.60? I think you just used the wrong henkaku version for your firmware. Try to uninstall enso and henkaku and reset your vita, then make sure to use the correct steps for your firmware.
> 
> ARK-4 works on 3.74, it's been tested before. The errors you're getting are not related to ARK but a bad installation of the hacks.

3.60, like I always said. Then why am I being said to not use 3.74 but 3.60? I already did that. One of my friends just did the same steps as me and he also had this problem so now there'w two users with this problem. I don't get it why I didn't have this problem before 4.20.68 Rev 2 when I followed the exact same steps.

krazynez, 2024-02-08:
> > So is your vita on 3.74 or 3.60? I think you just used the wrong henkaku version for your firmware. Try to uninstall enso and henkaku and reset your vita, then make sure to use the correct steps for your firmware.
> > ARK-4 works on 3.74, it's been tested before. The errors you're getting are not related to ARK but a bad installation of the hacks.
> 
> 3.60, like I always said. Then why am I being said to not use 3.74 but 3.60? I already did that. One of my friends just did the same steps as me and he also had this problem so now there's two users with this problem. I don't get it why I didn't have this problem before 4.20.68 Rev 2 when I followed the exact same steps.

We already went over this anything over 3.65 plugins are going to be an issue. FasterARK requires 2 plugin (same plugin but one is user and the other is kernel) and ARK-X requires another one.

JoseAaronLopezGarcia, 2024-02-08:
I don't think the issue is the plugins, they should work fine on 3.74 (I think). I believe he has installed the wrong henkaku version. Since he says he's on 3.60 yet he's never used henkaku.xyz, which is the correct henkaku version for 3.60, I don't know about this VitaDeploy app he's mentioning, I think that's the culprit here.

JoseAaronLopezGarcia, 2024-02-08:
Didn't you just say you're on 3.60?

Databoi2023, 2024-02-08:
> Didn't you just say you're on 3.60?
> 
> I've never used VitaDeploy before because installing henkaku via the official website is easy and works well.

Yes, I did, but you said to not downgrade using the steps I followed so how can I downgrade from 3.74, even though it's already downgraded to 3.60 by following the steps you say to not follow?

I don't know, never tried it. When I used the PS Vita for the first time, it already came with 3.73.

nallwolf, 2024-02-08:
Here's some clarification for everyone:

* VitaDeploy downgrades using modoru, there's no need to install the separate modoru
* The Vita that's having the issue is currently on 3.60 with enso
* HENlo was only used on 3.74 to initially hack the Vita, then it was downgraded to 3.60 with enso pre-installed, so there was never any reason to use henkaku.xyz
* Installing ARK should never mess with Vita homebrew

I'm going to get a Vita set up in the same way to see if I can replicate the issue.

Databoi2023, 2024-02-08:
> Here's some clarification for everyone:
>
> * VitaDeploy downgrades using modoru, there's no need to install the separate modoru
> * The Vita that's having the issue is currently on 3.60 with enso
> * [...]

Just to confirm the steps:

I was on 3.74, used the HENlo exploit, ran VitaDeploy, in "Install a different OS", I chose "Download 3.60 PUP" in "Target firmware", chose "HEN only" in "taiHEN configuration", checked "Install enso", clicked "Install the selected firmware". 

Afterwards, I used VitaShell's "App downloader" to install VitaShell. I uploaded AutoPlugin II from the PC to the PS Vita using FileZilla Client and VitaShell, installed it, ran it and installed the NoPspEmuDrm plugin from it. 

Then, I uploaded FasterARK from the PC to the PS Vita using FileZilla Client and VitaShell, installed it, ran it and installed ARK-4. 

Before even opening ARK-4, I ran VitaShell, a message with a "C2-12828-1" error appears and after clicking "OK", the PS Vita freezes with a black screen and I'm forced to access Safe Mode. If I use it to restart the PS Vita or rebuild its console, it freezes on spinning white dots, so I'm forced to access Safe Mode again and restore the PS Vita.

nallwolf, 2024-02-08:
It can be replicated. This was a Vita I updated to 3.74 and wiped before following all steps listed above to get on 3.60 enso with ARK. Can confirm it's broken in a way that it won't boot unless you hold L to disable plugins or wipe.

Problem seems to be caused by an issue in the way FasterARK edits the config.txt file. It doesn't insert a newline at the end of the file before adding the ps1cfw_enabler header so it ends up with "ur0:tai/NoPspEmuDrm_user.suprx*SCPS10084" as a line and tai really doesn't like that apparently. Booting with L and putting a newline where it was supposed to be fixes the issue and everything is now working as expected.

JoseAaronLopezGarcia, 2024-02-08:
Great findings @nallwolf! We'll fix it asap.

Databoi2023, 2024-02-08:
> It can be replicated. This was a Vita I updated to 3.74 and wiped before following all steps listed above to get on 3.60 enso with ARK. Can confirm it's broken in a way that it won't boot unless you [...]

I'm glad I'm not the only one having this problem. 

How did you put the newline?

nallwolf, 2024-02-08:
> I'm glad I'm not the only one having this problem.
>
> How did you put the newline?

After installing FasterARK, force power off and hold L while powering on. This makes taiHEN skip loading plugins, allowing you to launch VitaShell and edit ur0:tai/config.txt to add the newline then reboot and you're good to go!

Databoi2023, 2024-02-08:
> After installing FasterARK, force power off and hold L while powering on. This makes taiHEN skip loading plugins, allowing you to launch VitaShell and edit ur0:tai/config.txt to add the newline then [...]

I wish I new that before restoring the PS Vita!

JoseAaronLopezGarcia, 2024-02-08:
This pre-release should fix the issue, let me know if so and I'll update the main release with a new revision:
https://github.com/PSP-Archive/ARK-4/releases/tag/v1707435248

nallwolf, 2024-02-08:
> This pre-release should fix the issue, let me know if so and I'll update the main release with a new revision: https://github.com/PSP-Archive/ARK-4/releases/tag/v1707435248

This one is working as expected

JoseAaronLopezGarcia, 2024-02-08:
Alright, updated main release. I'm closing this issue, unless @Databoi2023 says otherwise and wants to reopen.

closed 2024-02-08 by JoseAaronLopezGarcia

Databoi2023, 2024-02-08:
> Alright, updated main release. I'm closing this issue, unless @Databoi2023 says otherwise and wants to reopen.

Thanks, it's working. Will you add an option to not install ARK-X in the future? 

Also, after installing ARK-4 via FasterARK and running it, ARK-4 sometimes freezes right away in the "ChovySign" part, and when running PSP ISOs, it sometimes has a black screen for some seconds until it crashes with the C1-2858-3 kernel error. I only have the previous issue if ARK-4's "Memory Stick Speedup" setting is set to "Always". Should I create a new issue regarding this problem?

JoseAaronLopezGarcia, 2024-02-08:
Just disable memory stick speedup. It's not really noticeable for the most part.

Databoi2023, 2024-02-09:
> Just disable memory stick speedup. It's not really noticeable for the most part.

Since I have this issue with that setting set to "Always", I think it should be set to "Disabled" as default.

Databoi2023, 2024-02-09:
> Just disable memory stick speedup. It's not really noticeable for the most part.

PSP ARK-4's updater is installing 4.20.68 Rev 3 instead of 4.20.68 Rev 4. The currently installed version was 4.20.68 Rev 3. I had to install 4.20.68 Rev 4 manually.

JoseAaronLopezGarcia, 2024-02-09:
It takes a while for servers to update. Try again.

Databoi2023, 2024-02-09:
> It takes a while for servers to update. Try again.

I already manually updated it but I'm having this issue for quite some time, even when I try to update ARK-4 days after the latest version was released.

Databoi2023, 2024-02-09:
> After installing FasterARK, force power off and hold L while powering on. This makes taiHEN skip loading plugins, allowing you to launch VitaShell and edit ur0:tai/config.txt to add the newline then [...]

Now I notice you mentioned in the previous comment to boot the PS Vita while holding the "L" button. It seems it boots the console with no plugins enabled.

Databoi2023, 2024-02-09:
Is a plugin like npdrm_free/npdrm_free_mod needed to read PSP EBOOT.PBP games and DLC, and if so, what about nploader/nploader_mod? Is it also needed?

Databoi2023, 2024-02-09:
> It takes a while for servers to update. Try again.

Is a plugin like npdrm_free/npdrm_free_mod needed to read PSP EBOOT.PBP games and DLC, and if so, what about nploader/nploader_mod? Is it also needed?

Also, what's PSP's ARK-4's default "Inferno Cache" setting, "LRU"?