PSP-Archive/ARK-4 #307: PS Vita's standalone ARK-4 crashes with the C1-2749-2 error after a clean ARK-4 install by FasterARK

Databoi2023, 2024-02-10:
### PSP/Vita Model

VTE-1000 (Vita TV)

### ARK Version

Version 4.20.68 Rev 7

### Steps to reproduce

The PS Vita's standalone ARK-4 crashes right away with the C1-2749-2 error after a clean ARK-4 install by FasterARK.

Also, ARK-X isn't installed (which I don't want, but just to let you know), only ARK-4.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

closed 2024-02-10 by Databoi2023

krazynez, 2024-02-10:
https://gbatemp.net/data/video/418/418628-c8473cbab78af1d73226dee9d36f640f.webm

Databoi2023, 2024-02-10:
> https://gbatemp.net/data/video/418/418628-c8473cbab78af1d73226dee9d36f640f.webm

I know. This time it was my fault.