PSP-Archive/ARK-4 #310: Some games don't work when "Unlock Extra Memory" is enabled.

Twinbeeman, 2024-02-13:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.67 r5

### Homebrew/GAME (if applicable)

ULES-00643, ULES-00939, ULES-00507, ULES-00664, ULES-00366, ULES-00813, ULES-01174, ULES-01226

### Steps to reproduce

Call of Duty /                             ULES-00643 -> Black screen after booting + crash
Indiana Jones /                          ULES-00939 -> Crashes at copyright screen
Open Season /                          ULES-00507 -> Black screen + crash after selecting new game
Over the Hedge /                      ULES-00664 -> Black screen after booting + crash
Pirates of the Caribbean DMC / ULES-00366 -> Green Screen + crash after selecting 1 Player mode
Shrek: the Third /                       ULES-00813 -> Crashes at copyright screen
Spiderman: Web of Shadows /    ULES-01174 -> Crashes at copyright screen
Xmen Origins: Wolverine /         ULES-01226 -> Crashes at first loading screen

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-02-13:
This is not an issue with the CFW. Absolutely no retail PSP game was made with the extra RAM in mind, so forcing it has unknown consequences. It works for most games, but it's not a guarantee for all of them.
The setting should only be used on games that are compatible and only when it's necessary (i.e. when using plugins like PRO Online or GTA Cheat Device Remastered).

closed 2024-02-13 by JoseAaronLopezGarcia

Twinbeeman, 2024-02-13:
and here I thought I was helping, lol, my bad. Is it best to change it to homebrew otherwise or should I just disable it altogether?

JoseAaronLopezGarcia, 2024-02-13:
It should be fine if you set it to homebrew.

Twinbeeman, 2024-02-13:
Alright, will do, thanks for the help and for all the work so far.