PSP-Archive/ARK-4 #311: Controller not working on ARK-X, Vita TV

iAmGhost, 2024-02-13:
### PSP/Vita Model

VTE-1000 (Vita TV)

### ARK Version

4.20.68 Rev 8

### Steps to reproduce

Vita version: 3.65
Controller: DualShock 4

[image]
ARK-X boots but controller does not working, also It has incorrectly cropped 4:3 screen.

* Installed ARK-X with FasterARK.vpk
* Tried disabling all plugins except nonpdrm, NoPspEmuDrm, ps1cfw_enabler, yamt.
* ARK(Standalone) works fine.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-02-13:
The 4:3 screen is how the PS1 image is, it can be changed to full screen in the settings by holding the PS button.
As for the controller thing, I'm not sure, I haven't actually used my Vita TV for a while, I'll try to test this when I get the chance.

JoseAaronLopezGarcia, 2024-02-17:
This one should fix it: https://github.com/PSP-Archive/ARK-4/releases/tag/v1708169291
It requires updating `ps1cfw_enabler` by either using FasterARK or by simply copying `PSVita/PS1CFW/ps1cfw_enabler.suprx` to `ur0:tai/`.

There's still one issue left to solve: the exit to launcher key combo (L1+R1+Start+Down on Vita TV) causes it to enter an infinite black screen and the console stops responding, so for the time being if you wanna close a game and open a new one you have to fully close and reopen ARK-X. I'm still trying to figure out how to fix this issue.

iAmGhost, 2024-02-17:
I can confirm controller works by replacing `ps1cfw_enabler.suprx` with [v1708169291](https://github.com/PSP-Archive/ARK-4/releases/tag/v1708169291).

FasterARK.vpk in that version were very buggy, It extracts content of savedata to `ux0:pspemu/PSP/SAVEDATA/` instead of `ux0:pspemu/PSP/SAVEDATA/ARK_01234`.

I found some other issues while testing, I'll submit them into separate issues.

JoseAaronLopezGarcia, 2024-02-17:
I fixed the exit to launcher key combo (L1+R1+Start+Down) in this pre-release: https://github.com/PSP-Archive/ARK-4/releases/tag/v1708174931

You can install it regularly (by running the UPDATE folder), no need to change the ps1cfw_enabler plugin (don't use FasterARK either, we need to fix that bug you just mentioned, which we thought we had fixed....)

closed 2024-02-17 by JoseAaronLopezGarcia

iAmGhost, 2024-02-17:
I found that exit to launcher combo does not working [v1708174931](https://github.com/PSP-Archive/ARK-4/releases/tag/v1708174931).
I think combination is not working. maybe because I'm using DualShock 4?

JoseAaronLopezGarcia, 2024-02-17:
Try latest release: https://github.com/PSP-Archive/ARK-4/releases/tag/v1708178401

Make sure to manually copy `ps1cfw_enabler.suprx` to `ur0:tai/` and `ARK_01234` to `ux0:pspemu/PSP/SAVEDATA/`.

Also from my experience sometimes the combo just doesn't want to work (and neither does cwcheat). Exiting from `ARK-X` and launching it again solves the issue.

krazynez, 2024-02-17:
> I can confirm controller works by replacing `ps1cfw_enabler.suprx` with [v1708169291](https://github.com/PSP-Archive/ARK-4/releases/tag/v1708169291).
>
> FasterARK.vpk in that version were very buggy, [...]

https://github.com/PSP-Archive/ARK-4/releases/tag/v1708179305

fixes this now also