PSP-Archive/ARK-4 #313: How to launch PRO ONLINE?

isMMa, 2024-02-14:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

6.60 ARK-4.20.68 r8 cIPL

### Homebrew/GAME (if applicable)

Alls games

### Plugins ( if applicable )

atPro on

### Steps to reproduce

Install atpro, configure server, configure ssid connection.

I set it up on another PSP with 6.60 ME.

PSP ARK4:
I start Tekken 6, I put lan and the connection does not turn green.

PSP 6.60 ME:
It connects in green to my router and searches for games.

PSP ARK4: PSP FAT 1000 1G
PSP 6.60 ME: PSP Slim 2000 2G

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-02-14:
PRO Online is not compatible with PSP 1K, you can use the DNS method instead.

isMMa, 2024-02-14:
> PRO Online is not compatible with PSP 1K, you can use the DNS method instead.

What is DNS method?

JoseAaronLopezGarcia, 2024-02-14:
https://psrewired.com/guides/psp

krazynez, 2024-02-14:
> https://psrewired.com/guides/psp

Seems that Tekken only supports adhoc not infrastructure mode.

closed 2024-02-17 by JoseAaronLopezGarcia