PSP-Archive/ARK-4 #314: Startup time is too slow on 1T SD.

gilgilgan, 2024-02-15:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

Version 4.20.68 Rev 8 (ARK-X)

### Steps to reproduce

Ever since I switched from Samsung 512G SD to Sandisk 1T SD, ARK is very slow to start.

t lags at the chovy project logo for close to 2 minutes. The SD continues to load during that time.

The SD is not a fake and I've used about 800G of it now, but I don't see any particular issues. 

What is unusual is that with the new ARK-X, it finishes loading just as quickly as it did when I was using 512G.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-02-15:
If there's too many games installed it will take a while to scan them all. Perhaps instead of using the main Game app, use the File Browser instead to navigate and launch your games. Organize your games in folders too to make it even faster to navigate. You can set the File Browser as the main app in the Menu Settings.

gilgilgan, 2024-02-16:
The delayed location is the screen where the #chovyProject logo is being printed. It has nothing to do with the ark explorer. I changed the initial location to the file explorer, but to no avail.

LiEnby, 2024-02-18:
> If there's too many games installed it will take a while to scan them all. Perhaps instead of using the main Game app, use the File Browser instead to navigate and launch your games. Organize your [...]

this is true, NoPspEmuDrm works by scanning the entirety of ux0:/pspemu/PSP/GAME and opening every .EDAT and .PBP file until one matching the content id of the rif file being opened is found, so the time it takes for the first time, will scale with how many PSP games you have ..

LiEnby, 2024-02-18:
> The delayed location is the screen where the #chovyProject logo is being printed. It has nothing to do with the ark explorer. I changed the initial location to the file explorer, but to no avail.

NoPspEmuDrm works by recursively scanning the entirety of ux0:/pspemu/PSP/GAME and opening every .EDAT and .PBP file until one matching the content id of the rif file being opened is found, its actually rather slow and the time it takes increases linearly with how many PSP games you have, it has a time complexity of o(n) - i would like to find a better method :d

the thing is, once the game is found it will write a fake .RIF file to your PSP/LICENSE folder that looks exactly like a real PSP rif file (except without a signature) and will just use that next time, this won't happen if there is already an offical (sony) RIF file for the game present however, in which it search for the PBP file every time.