PSP-Archive/ARK-4 #315: Missing some options in ARK-X, Vita TV

iAmGhost, 2024-02-17:
### PSP/Vita Model

VTE-1000 (Vita TV)

### ARK Version

4.20.68 Rev 8

### Steps to reproduce

I'm not sure this is ARK-X's problem or Vita FW's, but some options in Settings of ARK-X have fewer options than Vita(PCH-2000).

Controller Settings:
[image]
[image]

Other Settings:
[image]
[image]

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

nallwolf, 2024-02-17:
This is a normal limitation of the Vita TV. The menus are like this even in official PSN games.

1. There's no reason to assign buttons when the DS3/DS4 matches the DS1, only useful setting is to map left analog to d-pad
2. There are no touch panels to assign on the Vita TV
3. "Assign Controllers" is done on the first screen you get after holding the PS button

As for the other settings (screen size, volume, bilinear filtering), Sony restricted a lot on the Vita TV for no good reason, these are just another casualty.

iAmGhost, 2024-02-17:
> This is a normal limitation of the Vita TV. The menus are like this even in official PSN games.

I just found that too, thanks.
I found that I can enable those options by installing PSOneScrot plugin.
Since this is unrelated with ARK-X, I'll close this issue.

closed 2024-02-17 by iAmGhost